eloquentarduino/EloquentTinyML #3: tinymlgen

Workshopshed, 2020-05-13:
I used TinyMLGen to create a header file to incorporate into EloquentTinyML and it created my model as

`const unsigned char model_data[] DATA_ALIGN_ATTRIBUTE = ....`

When I compile I get the following error:

TextClassifierExample:13:95: error: invalid conversion from 'const unsigned char*' to 'unsigned char*' [-fpermissive]

Of course, I can simply remove the "const" but on the Arduino ARM boards the const keyword is also used to put the code into flash rather than ram, similar to the PROGMEM keyword on the AVR boards.

The solution seems to be to cast the variable but I was wondering if theree was a better way.

`Eloquent::TinyML::TfLite<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml((unsigned char*)model_data);`

eloquentarduino, 2020-05-13:
Unfortunately I don't have an ARM board. Will try to get one at soonest and check this issue. I will leave it open so I remember to come back to this.

eloquentarduino, 2021-10-31:
API changed, now you use `ml.begin(model_data)`.
Closing.

closed 2021-10-31 by eloquentarduino