eloquentarduino/EloquentTinyML #5: Possible issue with multi-class models.

MatInGit, 2020-05-17:
Hi,

I tried to deploy a 30 input to a 52 class output model using your library (like the frictionless approach btw). All classes come out as 0.00. I have retrained the models, made sure the scales are within float values range, and still no success. Running it on an ESP32. 

I have also tried a more minimal 2-in, 2-out model, which did not work either.

Any suggestions on what else I could try or is this an issue with the library?

eloquentarduino, 2020-05-17:
Will look further into this. I fear I fried one of my MCUs flashing a Tf model on it.

MatInGit, 2020-05-17:
I investigated further, turns out that TF Lite for MCU's does not yet support LSTM and sigmoid cells, which were part of the model I had uploaded. Thanks for the help anyway.

closed 2020-05-17 by MatInGit

whubaichuan, 2022-09-01:
@MatInGit @eloquentarduino Hi, thanks for your question and possible solution. But recently I came up with the same problem, i.e., All classes come out as 0.00. My EloquentTinyML version is 2.4.2.

Actually, I want to train the network by myself to get the .h file (my own weights) and test it in the digitsexample. I use your .py file listed below and test it using this [.ion project](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/DigitsExample.ino). But All classes come out as 0.00. 

`import math
import numpy as np
from sklearn.datasets import load_digits
import tensorflow as tf
from tensorflow.keras import layers
from tinymlgen import port

def get_data():
    np.random.seed(1337)
    x_values, y_values = load_digits(return_X_y=True)
    x_values /= x_values.max()
    # reshape to (8 x 8 x 1)
    x_values = x_values.reshape((len(x_values), 8, 8, 1))

    # split into train, validation, test
    TRAIN_SPLIT = int(0.6 * len(x_values))
    TEST_SPLIT = int(0.2 * len(x_values) + TRAIN_SPLIT)
    x_train, x_test, x_validate = np.split(x_values, [TRAIN_SPLIT, TEST_SPLIT])
    y_train, y_test, y_validate = np.split(y_values, [TRAIN_SPLIT, TEST_SPLIT])

    return x_train, x_test, x_validate, y_train, y_test, y_validate

def get_model():
    x_train, x_test, x_validate, y_train, y_test, y_validate = get_data()

    # create a CNN
    model = tf.keras.Sequential()
    model.add(layers.Conv2D(8, (3, 3), activation='relu', input_shape=(8, 8, 1)))
    # model.add(layers.MaxPooling2D((2, 2)))
    # model.add(layers.Conv2D(64, (3, 3), activation='relu'))
    # model.add(layers.MaxPooling2D((2, 2)))
    # model.add(layers.Conv2D(64, (3, 3), activation='relu'))
    model.add(layers.Flatten())
    # model.add(layers.Dense(16, activation='relu'))
    model.add(layers.Dense(len(np.unique(y_train))))

    model.compile(optimizer='adam', loss=tf.keras.losses.SparseCategoricalCrossentropy(from_logits=True), metrics=['accuracy'])
    model.fit(x_train, y_train, epochs=50, batch_size=16,
              validation_data=(x_validate, y_validate))
    return Exp

def test_model(model, x_test, y_test):
    x_test = (x_test / x_test.max()).reshape((len(x_test), 8, 8, 1))
    y_pred = model.predict(x_test).argmax(axis=1)
    print('ACCURACY', (y_pred == y_test).sum() / len(y_test))
    exit()

if __name__ == '__main__':
    model, x_test, y_test = get_model()
    test_model(model, x_test, y_test)
    c_code = port(model, variable_name='digits_model', pretty_print=True)
    print(c_code)`

Due to the version problem, I changed a little bit in this [.ion project](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/DigitsExample.ino) to run the code successfully when my EloquentTinyML version is 2.4.2.

from

 `#define NUMBER_OF_INPUTS 64
#define NUMBER_OF_OUTPUTS 10
// in future projects you may need to tweak this value: it's a trial and error process
#define TENSOR_ARENA_SIZE 8*1024

Eloquent::TinyML::TfLite<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml;`

to 

`
#define N_INPUTS 64
#define N_OUTPUTS 10
// in future projects you may need to tweak this value: it's a trial and error process
#define TENSOR_ARENA_SIZE 200*1024

Eloquent::TinyML::TensorFlow::TensorFlow<N_INPUTS, N_OUTPUTS, TENSOR_ARENA_SIZE> tf;

`

If I use the [.h file](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/digits_model.h) you provide, it's ok. Another try is that I construct the fully dense layers but also fail to all 0.0. I don't know what's the problem. Because the basic layer such as Conv2D, Dense should be supported by [tflite-micro](https://github.com/tensorflow/tflite-micro).

1. I try to re-train the [sine project](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/SineExample/sine_model.py) and this works on testing. But I failed in [digitsexample](https://github.com/eloquentarduino/EloquentTinyML/tree/master/examples/DigitsExample). Do you have any ideas? Could you share the correct .py file for digitsexample?
2. Is this because of the EloquentTinyML version? I am using 2.4.2.
3. I have double-checked the TENSOR_ARENA_SIZE and set it to the maximum  200*1024. Do you think is it the problem of TENSOR_ARENA_SIZE? I am using ARDUINO 33 BLE Sensor.

Looking for your reply.

Thanks a lot.

eloquentarduino, 2022-09-01:
@whubaichuan please share the Python file you're using, along with your TensorFlow version. I will take a look at this during the weekend.

whubaichuan, 2022-09-01:
> Environment (colab):
 tf. version  = 2.8.2
   tf.keras version = 2.8.0

The generation .h file python code of digitsexample is:

> I refer to this [.py file ](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/digits_model.py)

1. For input ( [64] ): 

```

import math
import numpy as np
from sklearn.datasets import load_digits
import tensorflow as tf
from tensorflow.keras import layers
from tinymlgen import port

def get_data():
    np.random.seed(1337)
    x_values, y_values = load_digits(return_X_y=True)
    #print('x_value_max ='+str(x_values.max()))
    x_values /= x_values.max()
    # reshape to (8 x 8 x 1)
    #x_values = x_values.reshape((len(x_values), 8, 8, 1))

    # split into train, validation, test
    TRAIN_SPLIT = int(0.6 * len(x_values))
    TEST_SPLIT = int(0.2 * len(x_values) + TRAIN_SPLIT)
    x_train, x_test, x_validate = np.split(x_values, [TRAIN_SPLIT, TEST_SPLIT])
    y_train, y_test, y_validate = np.split(y_values, [TRAIN_SPLIT, TEST_SPLIT])

    return x_train, x_test, x_validate, y_train, y_test, y_validate

def get_model():
    x_train, x_test, x_validate, y_train, y_test, y_validate = get_data()
    print(x_train.shape)

    model = tf.keras.Sequential()
    model.add(layers.Dense(32, activation='relu',input_shape=(64,)))
    #model.add(layers.BatchNormalization())
    #model.add(layers.Dropout(0.3))
    model.add(layers.Dense(16, activation='relu'))
    #model.add(layers.BatchNormalization())
    #model.add(layers.Dropout(0.3))
    #model.add(layers.Dense(len(np.unique(y_train),activation='softmax')))
    model.add(layers.Dense(10))
    #model.add(layers.Dense(10))

    model.compile(optimizer='adam', loss=tf.keras.losses.SparseCategoricalCrossentropy(from_logits=True), metrics=['accuracy'])
    print(model.summary())
    model.fit(x_train, y_train, epochs=50, batch_size=16, validation_data=(x_validate, y_validate))
    return model,x_test,y_test

def test_model_2(model, x_test, y_test): #for input 64
    x_test = (x_test / x_test.max())
    y_pred = model.predict(x_test).argmax(axis=1)
    print('ACCURACY', (y_pred == y_test).sum() / len(y_test))
    exit()

model, x_test, y_test = get_model()
test_model_2(model, x_test, y_test)
c_code = port(model, variable_name='digits_model', pretty_print=True)
print(c_code)

```

2. For input [8, 8, 1]:

[... 119 lines cut ...]

    Serial.print("Test output is: ");
    Serial.println(y_test);
    Serial.print("Predicted proba are: ");

    for (int i = 0; i < 10; i++) {
        Serial.print(y_pred[i]);
        Serial.print(i == 9 ? '\n' : ',');
    }

    Serial.print("Predicted class is: ");
    Serial.println(tf.probaToClass(y_pred));
    Serial.print("Sanity check: ");
    Serial.println(tf.predictClass(x_test));

    delay(1000);
}

```

If I test **step 3** using the template [.h file weights ](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/digits_model.h), it works. But if I test **step 3**  using the .h file weights produced by **step 1** and **step 2**, it will fail always, as the below picture shows.

[image]

Thanks for your help and looking forward to your solution!

eloquentarduino, 2022-09-02:
Ok, I need to make things a bit clearer in the docs. You have to apply 2 fixes:

 1. use tf==2.2.3. This is the guaranteed to work. 2.3 *may* work. 2.4+ are supposed NOT to work
 2. add `optimize=False` to  the `port` call: `port(model, variable_name='digits_model', pretty_print=True, optimize=False)`

I'm in the process of integrating the `tinymlgen` library into my new `everywhereml` library, so I'll update the docs there.
(as a sidenote, you have an `exit()` in your Python script you have to remove)

whubaichuan, 2022-09-02:
@eloquentarduino Thanks for your reply. I also update the information [here](https://github.com/eloquentarduino/EloquentTinyML/issues/15#issuecomment-1235147068). 

I use tf. version = 2.8.2, tf.keras version = 2.8.0 and optimize=False. It works finally.