eloquentarduino/EloquentTinyML #6: library dont work with Flatten()

beetlea, 2020-05-24:
Hi. 
I start your framework on esp32. If i run all full connected layers in my model, its work. But if i run cnn with flatten and full connected -> begin exception from esp from full connected layer tensorflow library

call esp32
Decoding stack results
0x4008eab4: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008ece5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400dfd85: tflite::reference_integer_ops::FullyConnected(tflite::FullyConnectedParams const&, tflite::RuntimeShape const&, signed char const*, tflite::RuntimeShape const&, signed char const*, tflite::RuntimeShape const&, int const*, tflite::RuntimeShape const&, signed char*) at C:\Users\Антон\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/kernels/internal/reference/integer_ops/fully_connected.h line 62
0x400e0171: tflite::ops::micro::fully_connected::Eval(TfLiteContext*, TfLiteNode*) at C:\Users\Антон\Documents\Arduino\libraries\EloquentTinyML\src\tensorflow\lite\micro\kernels\fully_connected.cpp line 133
0x400e8e86: tflite::MicroInterpreter::MicroInterpreter(tflite::Model const*, tflite::OpResolver const&, unsigned char*, unsigned int, tflite::ErrorReporter*) at C:\Users\Антон\Documents\Arduino\libraries\EloquentTinyML\src\tensorflow\lite\micro\micro_interpreter.cpp line 60
0x400d1c90: neural(void*) at C:\Users\РђРЅС‚РѕРЅ\Documents\Arduino\detect/detect.ino line 69
0x4008b115: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

eloquentarduino, 2020-05-25:
This library is just a wrapper around the Tf C++ api. As I see from the Tf Github page, Flatten is not currently supported. From the docs "TensorFlow Lite for Microcontrollers currently supports a limited subset of operations, so not all model architectures are possible."

[List of supported operators and layers](https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/micro/kernels/all_ops_resolver.cc)

closed 2020-06-01 by eloquentarduino

whubaichuan, 2022-09-02:
@beetlea update the supported operation by TFLite Micro, [here](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/all_ops_resolver.cc)

eloquentarduino, 2022-09-02:
Thanks @whubaichuan, updated in my comment too.

eloquentarduino, 2022-09-02:
By the way, CNN indeed works on ESP32. Be sure you're using Tf==2.2.3

whubaichuan, 2022-11-04:
@beetlea hi, flatten is not supported in the [official TFlite library](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/all_ops_resolver.cc). But @eloquentarduino in your [digits example ](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/digits_model.py), I see that you use `model.add(layers.Flatten())` and it works. So wired!

eloquentarduino, 2022-11-04:
@whubaichuan It looks like that official list is not accurate. I've found also other layers not listed works. It's a matter of trial and error.