eloquentarduino/EloquentTinyML #8: Example not working out of the box

coolcodecding, 2020-10-06:
I tried to run the SineExample an got the following two errors while compiling:

```
error: no matching function for call to 'Eloquent::TinyML::TfLite<1u, 1u, 2048u>::TfLite()'
 Eloquent::TinyML::TfLite<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml;
```

```
error: 'class Eloquent::TinyML::TfLite<1u, 1u, 2048u>' has no member named 'begin'
     ml.begin(sine_model);

```

Changing from:
`Eloquent::TinyML::TfLite<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml;
`
to

`Eloquent::TinyML::TfLite<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml((unsigned char*)sine_model);`

and removing the line
`ml.begin(sine_model);
`
during in setup() makes the code run.

SimoneS93, 2020-10-06:
Did you installed the library via the Library Manager?
The "not working" code reflects the latest version of the library you can download from Github.
The "working" code is based on an older version.

coolcodecding, 2020-10-06:
I followed the README.md of this repo. The example sketch on platformio seems to work, as it doesn't include the 
`ml.begin(sine_model)`. Seems to be the older version.
I didn't try Arduino IDE at all.

closed 2020-10-06 by coolcodecding