eloquentarduino/EloquentTinyML #9: print multiple outputs

coolcodecding, 2020-10-06:
Could you put an example in the README of how to deal with multiple outputs?

I have a model with 2 outputs (softmax) and changed the following lines in the code:

`#define NUMBER_OF_OUTPUTS 2
`
```
float output[2] = {};
float predicted = ml.predict(input, output);
```
and
`Serial.println(predicted[1]);
` as I want to get the second output.

I get the following error:
```
error: invalid types 'float[int]' for array subscript
     Serial.println(predicted[1]);
                               ^
```

Thanks!

eloquentarduino, 2020-10-06:
I will do it

eloquentarduino, 2020-10-18:
Sorry to have not replied earlier, I totally forgot this issue. Try to print output[1]

eloquentarduino, 2020-10-21:
See #10

closed 2020-10-21 by eloquentarduino