eloquentarduino/EloquentTinyML #10: Problem with multiple outputs

alefelipeoliveira, 2020-10-18:
Thanks fot the great Repo!
I am trying to use the but fot multiple inputs (3) and outputs(2),
However, the output always colleted the same value for both outputs.
Example:
I changed the inputs, and for each new input sets, the pair of outputs is different from the previous one, but the pair is equal:

float input[3] = { 10, 20, 30 };
float output[2] = { 0 };

ml.predict(input, output);

**Serial Console Printed:**
x: 1.92 y: 1.92
x: 1.30 y: 1.30

PS.: This issue do not occured with the model in the Keras notebook.

Notebook Environment: Google Colab
Microcontroller: ESP32 - ESP-WROOM-32

Follow my code:
https://github.com/alefelipeoliveira/EloquentTinyML_TEST

eloquentarduino, 2020-10-18:
I will have to look at this. I honestly don't use Tensorflow, so I have to make some tests to understand your problem.

alefelipeoliveira, 2020-10-20:
Hello,

Maybe the problem is in the function 
`float predict(float *input, float *output = NULL)`
This functions only handle with index 0, but it is interesting to check the lenght of the array output before return:
`return this->output->data.f[0];`

`output_1->output->data.f[1]
output_n->output->data.f[n]`

eloquentarduino, 2020-10-21:
It is true that it *returns* only the first index, but that's just for convenience. The function will still copy the `output->data.f` array into the `output` array, however.

I plan to make my tests this afternoon, please be patient.

In the meantime, you can try to inject a print inside the library `predict` function.

```cpp
// line 125, before if (output != NULL)

for (int i = 0; i < outputSize; i++)
    Serial.println(this->output->data.f[i]);
```

Does it prints the two outputs correctly?

eloquentarduino, 2020-10-21:
I can confirm it works with multiple outputs. Please see the new example I uploaded.
[SineCosineExample](https://github.com/eloquentarduino/EloquentTinyML/tree/master/examples/SineCosineExample)

Also be sure to use the latest (from Github) version of the library.

alefelipeoliveira, 2020-10-21:
I updated the library, and tested with my multiple input/output model,
Now it is working fine, 
Thanks a lot for the support

closed 2020-10-21 by eloquentarduino