eloquentarduino/EloquentTinyML #12: NameError: name 'Exp' is not defined, digits_model.py

zeta0707, 2020-12-25:
zeta@changwhan-asus:~/Arduino/DigitsExample$ python3 digits_model.py
2020-12-25 11:20:39.928140: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN)to use the following CPU instructions in performance-critical operations:  SSE4.1 SSE4.2 AVX AVX2 FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2020-12-25 11:20:39.948926: I tensorflow/core/platform/profile_utils/cpu_utils.cc:104] CPU Frequency: 2099940000 Hz
2020-12-25 11:20:39.949262: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x56281223f3f0 initialized for platform Host (this does not guarantee that XLA will be used). Devices:
2020-12-25 11:20:39.949294: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Host, Default Version
Epoch 1/50
68/68 [==============================] - 0s 2ms/step - loss: 2.0999 - accuracy: 0.4814 - val_loss: 1.7846 - val_accuracy: 0.7944
Epoch 2/50
68/68 [==============================] - 0s 970us/step - loss: 1.3560 - accuracy: 0.8516 - val_loss: 1.0277 - val_accuracy: 0.8194
Epoch 3/50
68/68 [==============================] - 0s 934us/step - loss: 0.6727 - accuracy: 0.9045 - val_loss: 0.6596 - val_accuracy: 0.8500
Epoch 4/50
68/68 [==============================] - 0s 948us/step - loss: 0.3958 - accuracy: 0.9314 - val_loss: 0.5517 - val_accuracy: 0.8528
Epoch 5/50
68/68 [==============================] - 0s 956us/step - loss: 0.2817 - accuracy: 0.9462 - val_loss: 0.4983 - val_accuracy: 0.8556
Epoch 6/50
68/68 [==============================] - 0s 954us/step - loss: 0.2200 - accuracy: 0.9536 - val_loss: 0.4707 - val_accuracy: 0.8611
Epoch 7/50
68/68 [==============================] - 0s 950us/step - loss: 0.1884 - accuracy: 0.9573 - val_loss: 0.4583 - val_accuracy: 0.8556
Epoch 8/50
68/68 [==============================] - 0s 947us/step - loss: 0.1621 - accuracy: 0.9601 - val_loss: 0.4408 - val_accuracy: 0.8722
Epoch 9/50
68/68 [==============================] - 0s 953us/step - loss: 0.1392 - accuracy: 0.9657 - val_loss: 0.4489 - val_accuracy: 0.8694
Epoch 10/50
68/68 [==============================] - 0s 948us/step - loss: 0.1262 - accuracy: 0.9712 - val_loss: 0.4396 - val_accuracy: 0.8778
Epoch 11/50
68/68 [==============================] - 0s 952us/step - loss: 0.1164 - accuracy: 0.9712 - val_loss: 0.4338 - val_accuracy: 0.8750
Epoch 12/50
68/68 [==============================] - 0s 944us/step - loss: 0.1056 - accuracy: 0.9722 - val_loss: 0.4306 - val_accuracy: 0.8833
Epoch 13/50
68/68 [==============================] - 0s 945us/step - loss: 0.1004 - accuracy: 0.9750 - val_loss: 0.4332 - val_accuracy: 0.8861
Epoch 14/50
68/68 [==============================] - 0s 944us/step - loss: 0.0922 - accuracy: 0.9796 - val_loss: 0.4203 - val_accuracy: 0.8806
Epoch 15/50
68/68 [==============================] - 0s 941us/step - loss: 0.0855 - accuracy: 0.9814 - val_loss: 0.4213 - val_accuracy: 0.8806
Epoch 16/50
68/68 [==============================] - 0s 941us/step - loss: 0.0793 - accuracy: 0.9833 - val_loss: 0.4277 - val_accuracy: 0.8917
Epoch 17/50
68/68 [==============================] - 0s 948us/step - loss: 0.0748 - accuracy: 0.9814 - val_loss: 0.4249 - val_accuracy: 0.8806
Epoch 18/50
68/68 [==============================] - 0s 938us/step - loss: 0.0728 - accuracy: 0.9805 - val_loss: 0.4228 - val_accuracy: 0.8917
Epoch 19/50
68/68 [==============================] - 0s 923us/step - loss: 0.0681 - accuracy: 0.9833 - val_loss: 0.4260 - val_accuracy: 0.8917
Epoch 20/50
68/68 [==============================] - 0s 953us/step - loss: 0.0634 - accuracy: 0.9889 - val_loss: 0.4480 - val_accuracy: 0.8944
Epoch 21/50
68/68 [==============================] - 0s 971us/step - loss: 0.0608 - accuracy: 0.9861 - val_loss: 0.4291 - val_accuracy: 0.8917
Epoch 22/50
68/68 [==============================] - 0s 961us/step - loss: 0.0574 - accuracy: 0.9879 - val_loss: 0.4460 - val_accuracy: 0.8917
Epoch 23/50
68/68 [==============================] - 0s 972us/step - loss: 0.0535 - accuracy: 0.9889 - val_loss: 0.4339 - val_accuracy: 0.8889
Epoch 24/50
68/68 [==============================] - 0s 953us/step - loss: 0.0510 - accuracy: 0.9889 - val_loss: 0.4394 - val_accuracy: 0.8944
Epoch 25/50
68/68 [==============================] - 0s 959us/step - loss: 0.0490 - accuracy: 0.9898 - val_loss: 0.4456 - val_accuracy: 0.8833
Epoch 26/50
68/68 [==============================] - 0s 972us/step - loss: 0.0461 - accuracy: 0.9917 - val_loss: 0.4377 - val_accuracy: 0.8972
Epoch 27/50
68/68 [==============================] - 0s 964us/step - loss: 0.0427 - accuracy: 0.9926 - val_loss: 0.4262 - val_accuracy: 0.8917
Epoch 28/50
68/68 [==============================] - 0s 956us/step - loss: 0.0423 - accuracy: 0.9944 - val_loss: 0.4533 - val_accuracy: 0.8972
Epoch 29/50
68/68 [==============================] - 0s 972us/step - loss: 0.0413 - accuracy: 0.9935 - val_loss: 0.4503 - val_accuracy: 0.8972
Epoch 30/50
68/68 [==============================] - 0s 955us/step - loss: 0.0397 - accuracy: 0.9917 - val_loss: 0.4468 - val_accuracy: 0.8972
Epoch 31/50
68/68 [==============================] - 0s 954us/step - loss: 0.0370 - accuracy: 0.9926 - val_loss: 0.4655 - val_accuracy: 0.8944
Epoch 32/50
68/68 [==============================] - 0s 951us/step - loss: 0.0350 - accuracy: 0.9926 - val_loss: 0.4370 - val_accuracy: 0.8917
Epoch 33/50
68/68 [==============================] - 0s 956us/step - loss: 0.0338 - accuracy: 0.9944 - val_loss: 0.4615 - val_accuracy: 0.8972
Epoch 34/50
68/68 [==============================] - 0s 972us/step - loss: 0.0327 - accuracy: 0.9954 - val_loss: 0.4541 - val_accuracy: 0.8972
Epoch 35/50
68/68 [==============================] - 0s 938us/step - loss: 0.0318 - accuracy: 0.9935 - val_loss: 0.4485 - val_accuracy: 0.8972
Epoch 36/50
68/68 [==============================] - 0s 958us/step - loss: 0.0290 - accuracy: 0.9963 - val_loss: 0.4737 - val_accuracy: 0.8972
Epoch 37/50
68/68 [==============================] - 0s 954us/step - loss: 0.0271 - accuracy: 0.9963 - val_loss: 0.4676 - val_accuracy: 0.9000
Epoch 38/50
68/68 [==============================] - 0s 952us/step - loss: 0.0275 - accuracy: 0.9963 - val_loss: 0.4745 - val_accuracy: 0.9028
Epoch 39/50
68/68 [==============================] - 0s 963us/step - loss: 0.0246 - accuracy: 0.9972 - val_loss: 0.4777 - val_accuracy: 0.9028
Epoch 40/50
68/68 [==============================] - 0s 964us/step - loss: 0.0244 - accuracy: 0.9972 - val_loss: 0.4763 - val_accuracy: 0.9000
Epoch 41/50
68/68 [==============================] - 0s 953us/step - loss: 0.0249 - accuracy: 0.9963 - val_loss: 0.4775 - val_accuracy: 0.8972
Epoch 42/50
68/68 [==============================] - 0s 951us/step - loss: 0.0232 - accuracy: 0.9981 - val_loss: 0.4792 - val_accuracy: 0.9000
Epoch 43/50
68/68 [==============================] - 0s 985us/step - loss: 0.0225 - accuracy: 0.9954 - val_loss: 0.4840 - val_accuracy: 0.8972
Epoch 44/50
68/68 [==============================] - 0s 947us/step - loss: 0.0199 - accuracy: 0.9981 - val_loss: 0.5112 - val_accuracy: 0.8972
Epoch 45/50
68/68 [==============================] - 0s 961us/step - loss: 0.0196 - accuracy: 0.9981 - val_loss: 0.4782 - val_accuracy: 0.8944
Epoch 46/50
68/68 [==============================] - 0s 935us/step - loss: 0.0188 - accuracy: 0.9991 - val_loss: 0.5014 - val_accuracy: 0.9028
Epoch 47/50
68/68 [==============================] - 0s 964us/step - loss: 0.0191 - accuracy: 0.9981 - val_loss: 0.5121 - val_accuracy: 0.9056
Epoch 48/50
68/68 [==============================] - 0s 931us/step - loss: 0.0167 - accuracy: 0.9991 - val_loss: 0.4977 - val_accuracy: 0.8917
Epoch 49/50
68/68 [==============================] - 0s 951us/step - loss: 0.0169 - accuracy: 0.9981 - val_loss: 0.5203 - val_accuracy: 0.9056
Epoch 50/50
68/68 [==============================] - 0s 970us/step - loss: 0.0164 - accuracy: 0.9991 - val_loss: 0.4874 - val_accuracy: 0.8972
Traceback (most recent call last):
  File "digits_model.py", line 52, in <module>
    model, x_test, y_test = get_model()
  File "digits_model.py", line 41, in get_model
    return Exp
**NameError: name 'Exp' is not defined**
(base) zeta@changwhan-asus:~/Arduino/DigitsExample$ 

In the digits_model.py, "return Exp" seems to be an error.
    model.fit(x_train, y_train, epochs=50, batch_size=16,
              validation_data=(x_validate, y_validate))
    return Exp

zeta0707, 2020-12-25:
HI sisco0,

Your resolution didn't work for me.
From: 
return Exp
To:
return model

```
68/68 [==============================] - 0s 935us/step - loss: 0.0151 - accuracy: 0.9991 - val_loss: 0.5223 - val_accuracy: 0.9000
Traceback (most recent call last):
  File "digits_model.py", line 52, in <module>
    model, x_test, y_test = get_model()
TypeError: cannot unpack non-iterable Sequential object
```
But I don't see any problem with this change.
From: 
return Exp
To:
return model, x_test, y_test

```
Epoch 50/50
68/68 [==============================] - 0s 952us/step - loss: 0.0163 - accuracy: 0.9991 - val_loss: 0.5426 - val_accuracy: 0.9000
ACCURACY 0.9554317548746518

```
Actually I got hit from example on below site.
https://eloquentarduino.github.io/2020/11/tinyml-on-arduino-and-stm32-cnn-convolutional-neural-network-example/

Thanks!

zeta0707, 2020-12-25:
Close since issue is fixed.

`return model, x_test, y_test`

closed 2020-12-25 by zeta0707

whubaichuan, 2022-09-02:
@zeta0707 Hi, thanks for your solution. By the way, have you ever tried to train the [digits_model.py](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/digits_model.py) by yourself and test it on board using the [.ion project](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/DigitsExample.ino)?  It seems a problem like h[ere](https://github.com/eloquentarduino/EloquentTinyML/issues/5#issuecomment-1234230756).

PR #13 (2020-12-25): Fixed return value for digits example
Closes #12 
The return value was not correctly specified, as `Exp` was not the correct one. The variable `model` has been set as the return value in order to go along with the other examples return value.