eloquentarduino/EloquentTinyML #14: has no member named 'probaToClass'

zeta0707, 2020-12-25:
I tried to run DigitsExample on ESP32 board, but I got below class error.
I installed 
[image]

```
/home/zeta/Arduino/DigitsExample/DigitsExample.ino: In function 'void loop()':
DigitsExample:50:23: error: 'class Eloquent::TinyML::TfLite<64u, 10u, 8192u>' has no member named 'probaToClass'
     Serial.println(ml.probaToClass(y_pred));
                       ^
DigitsExample:52:23: error: 'class Eloquent::TinyML::TfLite<64u, 10u, 8192u>' has no member named 'predictClass'
     Serial.println(ml.predictClass(x_test));
                       ^
exit status 1
![Screenshot_20201225_133221](https://user-images.githubusercontent.com/10907395/103119186-acf9a980-46b5-11eb-840b-a94c0a6216a8.png)

'class Eloquent::TinyML::TfLite<64u, 10u, 8192u>' has no member named 'probaToClass'
```

eloquentarduino, 2020-12-25:
Please update to v0.0.5

zeta0707, 2020-12-25:
Hi Eloquent,

Yes, I updated to v0.0.5, then issue was gone.
Thanks a lot!

zeta0707, 2020-12-25:
Close since issue is fixed.

closed 2020-12-25 by zeta0707