eloquentarduino/EloquentTinyML #15: digits_model.py

rkuo2000, 2021-01-04:
I found that optimize should set False, otherwise the digits_model.h can not be run (ESP32 keep aborting)

The below fix for digits_model.py are verified on NodeMCU32S (ESP32) with your Arduino DigitsExample
    c_code = port(model, variable_name='digits_model', pretty_print=True, optimize=False)
    with open('digits_model.h', 'w') as f:
        print(c_code, file=f)

rkuo2000, 2021-01-04:
After set Optimize=False to convert the model to ccode,
I can also run 28x28 MNIST model on ESP32 now
[TinyML_MNIST](https://github.com/rkuo2000/arduino/tree/master/examples/TinyML_MNIST)

eloquentarduino, 2021-01-05:
Where did you find `optimize=True` in my tutorials? I already switched to `optimize=False` in the latest version of `tinymlgen`.

rkuo2000, 2021-01-05:
https://eloquentarduino.github.io/2020/01/easy-tinyml-on-esp32-and-arduino/
converter.optimizations = [tf.lite.Optimize.OPTIMIZE_FOR_SIZE]  <-- this indicates it needs OPTIMIZED_FOR_SIZE

and

c_code = port(model, pretty_print=True)  <-- also in your digits_model.py 
print(c_code)

The above code did not specify optimize=False, 
and it will generate a very small code. (my tinymlgen is v0.2 which is the current version)

eloquentarduino, 2021-01-05:
Thanks, I removed `converter.optimizations = [tf.lite.Optimize.OPTIMIZE_FOR_SIZE] ` from the post. 
`tinymlgen` version `0.3` switched to `optimize=False` by default

rkuo2000, 2021-01-05:
Thanks for your great work !

closed 2021-01-05 by rkuo2000

whubaichuan, 2022-09-02:
@eloquentarduino It seems that in 2.4.3 (the latest version currently on 2022.09.02), 

`c_code = port(model, pretty_print=True) <-- also in your digits_model.py
print(c_code)`

does not specify optimize=False,
and it will generate a very small code.

eloquentarduino, 2022-09-02:
Optimization seems to break the exported code. For the moment I will default to `optimize=False`