eloquentarduino/EloquentTinyML #16: Conv1D & MaxPooling1D

rkuo2000, 2021-01-05:
After I successfully using Kaggle to generate [TinyML MNIST](https://www.kaggle.com/rkuo2000/tinyml-mnist) and running on ESP32 using its Arduino [TinyML_MNIST](https://github.com/rkuo2000/arduino/tree/master/examples/TinyML_MNIST) sample code

I am creating another example which is [TinyML ECG](https://www.kaggle.com/rkuo2000/tinyml-ecg), 
it uses Conv1D & MaxPooling1D  for ECG signals, input_shape=(187, 1)
however, it is not successful to run on ESP32 using its Arduino [TinyML_ECG](https://github.com/rkuo2000/arduino/tree/master/examples/TinyML_ECG) sample code

I tried different TENSOR_ARENA_SIZE ,  32/64 *1024, still not able to get prediction result, 
the below message seems not running the model with x_test.h, y_pred are all 0.00 (this ECG num_classes = 5)
It took 3 micros to run inference
0.00,0.00,0.00,0.00,0.00

Any hint to debug this ?

eloquentarduino, 2021-01-05:
Can you please share a Notebook of this?

rkuo2000, 2021-01-06:
It is based on Kaggle https://www.kaggle.com/rkuo2000/tinyml-ecg
it will include the dataset https://www.kaggle.com/shayanfazeli/heartbeat
.ipynb is attached.

eloquentarduino, 2021-01-09:
I'm getting this error using your code.

```
GetModel done
Version check done
Didn't find op for builtin opcode 'EXPAND_DIMS' version '1'

Failed to get registration from op code  d
 

Allocation failed
AllocateTensors() failed

Cannot load Tf model
```

rkuo2000, 2021-01-09:
I have revised the [python code](https://www.kaggle.com/rkuo2000/tinyml-ecg) 
Version 1. with DNN , the arduino code can run fine to predict result on ESP32
Version 2. with Conv1D+Conv1D+Maxpooling1D+DNN, the arduino code can not predict
Version 3. with Conv1D+DNN, the arduino code can not predict
it seems Conv1D is not getting input data (187,1) ?!

rkuo2000, 2021-01-09:
I have revised the python code <https://www.kaggle.com/rkuo2000/tinyml-ecg>
Version 1. with DNN , the arduino code predict OK
Version 2. with Conv1D+Conv1D+Maxpooling1D+DNN, the arduino code did NOT
predict
Version 3. with Conv1D+DNN, the arduino code did NOT predict
Version 4. with Conv2D+DNN (need X_train.reshape(-1, 1, 187,1), predict OK
(more code size, 96.18% val accuracy, predict fast 44.1ms)
Version 5. with Conv2D+Conv2D+Maxpooling2D+DNN, predict OK (less code size,
97.2 val accuracy, but predict slower 265ms)

it seems Conv1D is not converted to C code correctly ?!

eloquentarduino, 2021-01-10:
Conv1D is not listed in their supported operators: https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/micro/all_ops_resolver.cc

rkuo2000, 2021-01-10:
Thanks for your help to clarify it, I have kept my [python example code](https://www.kaggle.com/rkuo2000/tinyml-ecg) in Conv2D.

closed 2021-01-10 by rkuo2000

AhmedElkhamisi, 2021-07-25:
@rkuo2000 Bro, I tried to use your code (2D CNN) but I'm getting a prediction of "nan" on the serial monitor. Can I send you my code if you can help me out?

rkuo2000, 2021-07-25:
Yes, of course you can.

rkuo2000, 2021-07-25:
It is probably your data issue, I have several Kaggle examples for  your
reference below:
https://www.kaggle.com/rkuo2000/tinyml-airdigit
https://www.kaggle.com/rkuo2000/tinyml-ecg
https://www.kaggle.com/rkuo2000/tinyml-mnist
https://www.kaggle.com/rkuo2000/tinyml-gtsrb
https://www.kaggle.com/rkuo2000/tinyml-digit

AhmedElkhamisi, 2021-07-25:
Okay thanks I'll check them out and if I faced any problem I'll send you the code.

AhmedElkhamisi, 2021-07-25:
Is the export test data essential?? My project is on ECG too.

AhmedElkhamisi, 2021-07-25:
How do you upload your code to arduino?

AhmedElkhamisi, 2021-07-25:
@rkuo2000 How do you upload your code to arduino? I used your exact same code but it didn't predict for me.

rkuo2000, 2021-07-25:
Please look into
https://github.com/rkuo2000/arduino/tree/master/examples
TinyML_AirDigit
TinyML_Digit
TinyML_ECG
TinyML_MNIST

upload code into ESP32 (NodeMCU-32S) using Arduino IDE

AhmedElkhamisi, 2021-07-25:
> Please look into https://github.com/rkuo2000/arduino/tree/master/examples TinyML_AirDigit TinyML_Digit TinyML_ECG TinyML_MNIST upload code into ESP32 (NodeMCU-32S) using Arduino IDE AhmedElkhamisi ***@***.***> 於 2021年7月26日 週一 上午3:17寫道：
> […](#)
> -- Richard Kuo (郭子仁) Adjunct Assistant Professor Electrical Engineering Dept., National Taiwan Ocean University +886-972211921

Thanks a lot!

eloquentarduino, 2021-07-26:
@AhmedElkhamisi did you solved your error? I get many requests of help for NaN values, I will definetly update the examples to clarify how the input should be formatted for Conv2D to work

@rkuo2000 I'm glad you now mastered the TinyML workflow

AhmedElkhamisi, 2021-07-26:
@eloquentarduino Yes I did! It was a problem with the Arduino code not the python code.

AhmedElkhamisi, 2021-07-26:
@rkuo2000 Thankyou once again!

whubaichuan, 2022-11-04:
@AhmedElkhamisi @rkuo2000 hi, do you have a similar problem like https://github.com/eloquentarduino/EloquentTinyML/issues/50 when transferring the EloquentTnyML examples to your own project? Thanks.

whubaichuan, 2022-11-04:
> did you solved your error? I get many requests of help for NaN values, I will definetly update the examples to clarify how the input should be formatted for Conv2D to work

@eloquentarduino hi, I am also facing this [problem](https://github.com/eloquentarduino/EloquentTinyML/issues/50). Could you update an example to clarify how the input should be formatted for Conv2D to work?

eloquentarduino, 2022-11-05:
Yes, I will create a more detailed guide for the different types of input data.