eloquentarduino/EloquentTinyML #18: Issues with DigitsModel example

Technerder, 2021-03-04:
The get_model method in `digits_model.py` does not return what it is supposed to (returns `Exp` as opposed to returning `model, x_test, and y_test`). Furthermore, even after modifying the script like shown below the code doesn't seem to work on any microcontrollers (after replacing the old digits_model.h with the new one generated by the script)

```py
import numpy as np
import tensorflow as tf
from tinymlgen import port
from tensorflow.keras import layers
from sklearn.datasets import load_digits

def get_data():
    np.random.seed(1337)
    x_values, y_values = load_digits(return_X_y=True)
    x_values /= x_values.max()
    # reshape to (8 x 8 x 1)
    x_values = x_values.reshape((len(x_values), 8, 8, 1))

    # split into train, validation, test
    TRAIN_SPLIT = int(0.6 * len(x_values))
    TEST_SPLIT = int(0.2 * len(x_values) + TRAIN_SPLIT)
    x_train, x_test, x_validate = np.split(x_values, [TRAIN_SPLIT, TEST_SPLIT])
    y_train, y_test, y_validate = np.split(y_values, [TRAIN_SPLIT, TEST_SPLIT])

    return x_train, x_test, x_validate, y_train, y_test, y_validate

def get_model():
    x_train, x_test, x_validate, y_train, y_test, y_validate = get_data()
    # create a CNN
    model = tf.keras.Sequential()
    model.add(layers.Conv2D(8, (3, 3), activation='relu', input_shape=(8, 8, 1)))
    # model.add(layers.MaxPooling2D((2, 2)))
    # model.add(layers.Conv2D(64, (3, 3), activation='relu'))
    # model.add(layers.MaxPooling2D((2, 2)))
    # model.add(layers.Conv2D(64, (3, 3), activation='relu'))
    model.add(layers.Flatten())
    # model.add(layers.Dense(16, activation='relu'))
    model.add(layers.Dense(len(np.unique(y_train))))
    model.compile(optimizer='adam', loss=tf.keras.losses.SparseCategoricalCrossentropy(from_logits=True), metrics=['accuracy'])
    model.fit(x_train, y_train, epochs=50, batch_size=16, validation_data=(x_validate, y_validate))
    return model, x_test, y_test

if __name__ == '__main__':
    model, x_test, y_test = get_model()
    open('selfdigits.h', 'w').write(port(model, variable_name='digits_model', pretty_print=True))
```

What am I doing wrong?

eloquentarduino, 2021-03-05:
Please take a look now. I re-committed the fixes.

Technerder, 2021-03-05:
The re-commit seems to have properly merged the push, however the model output provided by the script still doesn't seem to work properly on any microcontrollers (Teensy 4.0, NodeMCU 0.9) I'm testing them on, which is odd since the repository provided [model](https://raw.githubusercontent.com/eloquentarduino/EloquentTinyML/master/examples/DigitsExample/digits_model.h) works seamlessly, which doesn't make much sense to me.

eloquentarduino, 2021-03-05:
Try this

```cpp
open('selfdigits.h', 'w').write(port(model, variable_name='digits_model', pretty_print=True, optimize=False))
```

Technerder, 2021-03-05:
That seems to make the generated models work.

closed 2021-03-05 by eloquentarduino