eloquentarduino/EloquentTinyML #22: cannot allocate tensors with own generated model for Sine example

Vensim, 2021-03-31:
The output of the serial comms is. 
[image]

What I've tried : 
Increasing and decreasing Tensor arena size, still same error.

Haven't done much changes to the model generation other than outputting straight to a c file :
```cpp
# Port to C
c_code = port(model, pretty_print = True)
c_file = open("sine_model.h", "w")
c_file.write(c_code)
```

Here are my files and the model that's being used. 
Python script :
```python
import math
import numpy as np
import tensorflow as tf
from tensorflow.keras import layers
from tinymlgen import port

def get_model():
    SAMPLES = 1000
    np.random.seed(1337)
    x_values = np.random.uniform(low=0, high=2*math.pi, size=SAMPLES)
    # shuffle and add noise
    np.random.shuffle(x_values)
    y_values = np.sin(x_values)
    y_values += 0.1 * np.random.randn(*y_values.shape)

    # split into train, validation, test
    TRAIN_SPLIT =  int(0.6 * SAMPLES)
    TEST_SPLIT = int(0.2 * SAMPLES + TRAIN_SPLIT)
    x_train, x_test, x_validate = np.split(x_values, [TRAIN_SPLIT, TEST_SPLIT])
    y_train, y_test, y_validate = np.split(y_values, [TRAIN_SPLIT, TEST_SPLIT])

    # create a NN with 2 layers of 16 neurons
    model = tf.keras.Sequential()
    model.add(layers.Dense(8, activation='relu', input_shape=(1,)))
    model.add(layers.Dense(16, activation='relu'))
    model.add(layers.Dense(1))
    model.compile(optimizer='rmsprop', loss='mse', metrics=['mae'])
    model.fit(x_train, y_train, epochs=200, batch_size=16,
                        validation_data=(x_validate, y_validate))
    return model

def test_model(model, verbose=False):
    x_test = np.random.uniform(low=0, high=2*math.pi, size=100)
    y_test = np.sin(x_test)
    y_pred = model.predict(x_test)
    print('MAE', np.abs(y_pred - y_test).mean())

model = get_model()
converter = tf.lite.TFLiteConverter.from_keras_model(model)
converter.optimizations = [tf.lite.Optimize.OPTIMIZE_FOR_SIZE]
tflite_model = converter.convert()
test_model(model)

# Save the model to disk
open("sine_model_quantized.tflite", "wb").write(tflite_model)

# Port to C
c_code = port(model, pretty_print = True)
c_file = open("sine_model.h", "w")
c_file.write(c_code)
```
Arduino IDE code :
```cpp
[... 228 lines cut ...]
	0x61, 0x72, 0x69, 0x61, 0x62, 0x6c, 0x65, 0x4f, 0x70, 0x2f, 0x72, 0x65, 
	0x73, 0x6f, 0x75, 0x72, 0x63, 0x65, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 
	0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0e, 0x00, 0x14, 0x00, 0x04, 0x00, 
	0x00, 0x00, 0x08, 0x00, 0x0c, 0x00, 0x10, 0x00, 0x0e, 0x00, 0x00, 0x00, 
	0x4c, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00, 0x0c, 0x00, 0x00, 0x00, 
	0x04, 0x00, 0x00, 0x00, 0x84, 0xff, 0xff, 0xff, 0x30, 0x00, 0x00, 0x00, 
	0x73, 0x65, 0x71, 0x75, 0x65, 0x6e, 0x74, 0x69, 0x61, 0x6c, 0x2f, 0x64, 
	0x65, 0x6e, 0x73, 0x65, 0x2f, 0x42, 0x69, 0x61, 0x73, 0x41, 0x64, 0x64, 
	0x2f, 0x52, 0x65, 0x61, 0x64, 0x56, 0x61, 0x72, 0x69, 0x61, 0x62, 0x6c, 
	0x65, 0x4f, 0x70, 0x2f, 0x72, 0x65, 0x73, 0x6f, 0x75, 0x72, 0x63, 0x65, 
	0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 
	0x14, 0x00, 0x18, 0x00, 0x04, 0x00, 0x00, 0x00, 0x08, 0x00, 0x0c, 0x00, 
	0x10, 0x00, 0x00, 0x00, 0x00, 0x00, 0x14, 0x00, 0x14, 0x00, 0x00, 0x00, 
	0x38, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x20, 0x00, 0x00, 0x00, 
	0x18, 0x00, 0x00, 0x00, 0x04, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00, 
	0xff, 0xff, 0xff, 0xff, 0x01, 0x00, 0x00, 0x00, 0x04, 0x00, 0x04, 0x00, 
	0x04, 0x00, 0x00, 0x00, 0x0b, 0x00, 0x00, 0x00, 0x64, 0x65, 0x6e, 0x73, 
	0x65, 0x5f, 0x69, 0x6e, 0x70, 0x75, 0x74, 0x00, 0x02, 0x00, 0x00, 0x00, 
	0x01, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 
	0x10, 0x00, 0x00, 0x00, 0x0c, 0x00, 0x10, 0x00, 0x07, 0x00, 0x00, 0x00, 
	0x08, 0x00, 0x0c, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x09, 
	0x09, 0x00, 0x00, 0x00, 0x09, 0x00, 0x00, 0x00
};
const int model_data_len = 2384;
```

Vensim, 2021-03-31:
[files.zip](https://github.com/eloquentarduino/EloquentTinyML/files/6238607/files.zip)

Vensim, 2021-03-31:
Model output fixed upon inspection of Issue 17's comments.

> Replace
> 
> ```c++
> c_code = port(model, pretty_print=True)
> ```
> 
> with
> 
> ```c++
> c_code = port(model, pretty_print=True, optimize=False)
> ```

Sent request.

closed 2021-03-31 by Vensim

whubaichuan, 2022-09-02:
@eloquentarduino I suggest changing the code in the original file. It seems lots of people ask the same question including me at the beginning. Thanks.

Vensim, 2022-09-02:
@eloquentarduino Is this the same issue as was posted here?
I see on updated example of Sine , `optimize=False` is missing, did adding it fix it for you?

eloquentarduino, 2022-09-02:
Publishing EloquentTinyML==2.4.4 with `optimize=False` in every Python example. I will migrate the `tinymlgen.port` function into my new `everywhereml` package.