eloquentarduino/EloquentTinyML #24: int32 error compiling for stm32f103c series

cesarares, 2021-04-08:
compiling in Arduino IDE for stm32f103c series generate this error: libmaple_types.h:49:13: error: 'int32' has a previous declaration as 'typedef int int32'

eloquentarduino, 2021-04-11:
I cannot test because I don't have a board, but at least this fix makes the Sine example compile.

Go to `Arduino/libraries/EloquentTinyML/src/tensorflow/lite/kernels/internal/compatibility.h` line 84. Replace

```cpp
using int32 = std::int32_t;
```

with 

```cpp
#define int32 foo
using int32 = std::int32_t;
```

I won't publish this hack because I don't know if it can break somethings. Please report back your results.

cesarares, 2021-04-14:
We have a small project where we are ML algorithms in small microcontrollers. In particular we are testing a convolutional network model to recognize 8x8 digits. The sketch compiled fine for the stm32f103c8t6 (Arm Cortex-M3) development board but the code exceeded the rom by 26.6KiB. We ended up successfully implementing the functions in C to do the inference (we were surprised that our implementation required olmost half the ram and rom of this library).
We also tested the modification you proposed with an ESP32 development board and It compiled and it worked correctly.
I leave below some measurements that we made on the model and library in case they may be useful to you. 

Library                   |           Dev Board(MCU)         | Data Mem.  | Prog. Mem. | Inference Time | Accuracy
Eloquent TinyML   |      Xtensa LX6 (Esp32)           |    23.12 KiB  |    470.6 KiB |             2270 us |   97,80%
C Implementation |      Xtensa LX6 (Esp32)           |    14.21 KiB  |    277.1 KiB |               606 us |   97,80%
Eloquent TinyML   | Tensilica L106 (Esp8266)        |    51.20 KiB  |   391.5 KiB  |          11167 us |   97,80%
C Implementation | Tensilica L106 (Esp8266)        |    31.01 KiB  |   274.0 KiB  |            7568 us |   97,80%
C Implementation | Stm32f103  (Arm Cortex-M3) |      2.14 KiB |     27.3 KiB  |            8624 us |   97,80%

>

closed 2022-09-02 by eloquentarduino