eloquentarduino/EloquentTinyML #26: TinyML for arduino Uno

DeepthiSudharsan, 2021-05-25:
Hi!

I was going through this wonderful post of yours https://eloquentarduino.github.io/2021/03/tinyml-classification-example-wine-dataset/ .

I was trying to understand and replicate the part where you convert the code to C++ and run the neural network on the microcontroller. I tried doing that process with the wine dataset as in your blog and also just tried running the EloquentTinyMl examples on my Arduino Uno. But I am repeatedly getting this same error.

[image]

[image]

I am unable to figure out where I am going wrong and how to go about it. Any help on this issue is very much appreciated :).

Thank you for your wonderful articles on TinyMl once again!!

ankaiser4711, 2021-05-25:
arduino uno has to less memory to perform any tinyml stuff. 
try a ‚bigger’ controller. f. e. esp32. 
see benchmark posts for tested boards. 

---------------------------
Andreas Kaiser
Karl-Marx-Str. 19
D-67466 Lambrecht

***@***.***

www.swe-kaiser.de
www.scherwel.de
Tel. 0170 7723834
Fax. 03212 1078324

DeepthiSudharsan, 2021-05-25:
Oh that you very much for the info @ankaiser4711 ! Currently, I have only Raspberry pi and Arduino Uno at my disposal to work with. Will try finding an alternative way. Thanks a lot!

eloquentarduino, 2021-05-25:
@ankaiser4711 is correct. 2 Kb of RAM of the UNO is way under the minimum requirement for Tensorflow. I suggest you look at alternative algorithms, like [Decision Tree and Random Forest](https://eloquentarduino.github.io/2020/10/decision-tree-random-forest-and-xgboost-on-arduino/)

closed 2021-05-25 by eloquentarduino

DeepthiSudharsan, 2021-05-25:
Thank you so much @eloquentarduino . Will try it out!