eloquentarduino/EloquentTinyML #27: Not able to run the SineFromInternetExample.ino on Nano RP2040

KejaPower, 2021-06-06:
Hello,
I'm trying the SineFromInternetExample.ino example on my Nano RP2040 everything works fine until it tries to load the model : `ml.begin(model)`
 at that point the board is stuck. From the Serial Monitor I get : 

```
Model size is: 2280

Model content: 1C 0 0 0 54 46 4C 33 14 0 20 0 4 0 8 0 C 0 10 0  ... A6 C0 AD 6F DD AA 9 61 4A DD B0 B8 FF 51 3B A0 57 7E 7F 72 
```
 
so it seems to me I able to get the model from the server.

Any suggestions on what I m doing wrong ?

Thanks

eloquentarduino, 2021-06-07:
I had the same error with my Nano Rp2040. The first characters from the model are correct, the last ones no: it should end in 00 04 00 00 00.
Since the exact same code works on the ESP32, I believe it has something to do with the WiFiNINA library. Will keep you informed as soon as I solve this issue (I already have a fix in my that need to try).

KejaPower, 2021-06-08:
thanks for looking into this.
I have tried other http arduino librairies with the same results. It  indeed seems to point to a problem with the WifiNINA librairy.
I have an ESP32 board on the way but would be great to get it working on the RP2040.

eloquentarduino, 2021-06-08:
I fixed the bug, it was what I suspected. Check it out https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/SineFromInternetExample/SineFromInternetExample.ino
(only changed line 97)

closed 2021-06-08 by eloquentarduino

KejaPower, 2021-06-08:
thanks , working now !