eloquentarduino/EloquentTinyML #28: Using the library to predict a model with 2 features as an input (2d array) not just one

ahmedmazen191, 2021-07-04:
thanks for the wonderful effort to do such an effective and simple library. I was wondering if I can use the library to predict a model with 2 features as an input (2d array) not just one. for instance the X position and Y position of the peaks of an FFT as inputs to my model

eloquentarduino, 2021-07-05:
You have to reshape the 2d array to 1d as [X0, Y0, X1, Y1, X2, Y2, ...]

ahmedmazen191, 2021-07-05:
thanks A lot  I did that with a reshape layer in my model and it worked

closed 2021-07-05 by ahmedmazen191

whubaichuan, 2022-09-02:
@eloquentarduino @ahmedmazen191 It's amazing. I try to input the [64] image in .ino project for an [8x8x1] inputting Conv2D network, and it works. Maybe this library will change the input dimension automatically according to the .h file (C code). Magic!

whubaichuan, 2022-11-04:
@ahmedmazen191 hi, do you have a similar problem like [here](https://github.com/eloquentarduino/EloquentTinyML/issues/50) when transferring the EloquentTnyML examples to your own project? Thanks.