eloquentarduino/EloquentTinyML #29: Three spaces too much

maarten-pennings, 2021-08-30:
Thanks for this library. Worked immediately!

Small improvement suggestion for an error message on line 418 in  micro_allocator.cpp [link](https://github.com/eloquentarduino/EloquentTinyML/blob/bcec14efad4f696860a197d96818a32882e1f66c/src/tensorflow/lite/micro/micro_allocator.cpp#L418)

The printed string currently looks like this
```
"Failed to get registration from op code % d\n "
```

The space between `%` and `d` should go, but probably also the one  after `\n` and the one between `op` and `code`:

```
"Failed to get registration from opcode %d\n"
```

eloquentarduino, 2021-08-30:
Done, thank you for your feedback.
Version 0.0.10 in Arduino IDE should contain the fix.

closed 2021-08-30 by eloquentarduino

maarten-pennings, 2021-09-06:
I now get this error.

```
Didn't find op for builtin opcode 'CONV_2D' version '5'
Failed to get registration from op code 1061428504
```

I don't know the internals, but to me it looks like the `%d` now prints the _address_ of the `struct OperatorCode`.

eloquentarduino, 2021-09-06:
What you're pointing out here is out of scope of the library. EloquentTinyML is just a wrapper around Tensorflow for Microcontrollers: you should post the errors from the Tf library on the Tf repository, not here.
I really can't help because I hardly use myself Tf. Many users point out this kind of error: it often means you're using a Tf version in Python that is not supported on the MCU: this library contains a pre-built version of Tf 2.1. If you're using Tf 2.5 on your PC, try to switch to 2.1 and see if it works.

maarten-pennings, 2021-09-06:
Sorry. I was not referring to my "real" problem here: the unsupported opcode.
I was referring to the fact that the [error](https://github.com/eloquentarduino/EloquentTinyML/blob/master/src/tensorflow/lite/micro/micro_allocator.cpp#L418) message seems to print an address (`%d`) instead of stringifying the object `opcode` or is this also part of the wrapped "Tensorflow for Microcontrollers"?

eloquentarduino, 2021-09-07:
It is part of the Tensorflow for Microcontrollers library. I don't think you can get a stringified version of it, you already have 

```
Didn't find op for builtin opcode 'CONV_2D' version '5'
```

as a meaningful message.