eloquentarduino/EloquentTinyML #30: Error compiling for board DOIT ESP32 DEVKIT V1

embedragx, 2021-09-18:
In file included from c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\svdf.cpp:25:
c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src/tensorflow/lite/micro/kernels/activation_utils.h: In function 'float tflite::ops::micro::ActivationValFloat(TfLiteFusedActivation, float)':
c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src/tensorflow/lite/micro/kernels/activation_utils.h:53:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\svdf.cpp: In function 'TfLiteStatus tflite::ops::micro::svdf::Eval(TfLiteContext*, TfLiteNode*)':
c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\svdf.cpp:558:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 45 1406 1640 111 (set (reg:SF 19 f0 [orig:198 _248 ] [198])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC101") [flags 0x2]) [0  S4 A32])) "c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src/tensorflow/lite/micro/kernels/activation_utils.h":44 47 {movsf_internal}
     (expr_list:REG_EQUAL (const_double:SF 6.0e+0 [0x0.cp+3])
        (nil)))
during RTL pass: postreload
c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\svdf.cpp:558:1: internal compiler error: in extract_constrain_insn, at recog.c:2210
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
during RTL pass: ira
c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\unpack.cpp: In function 'TfLiteStatus tflite::ops::micro::unpack::{anonymous}::Eval(TfLiteContext*, TfLiteNode*)':
c:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\unpack.cpp:104:1: internal compiler error: Segmentation fault
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
Compilation error: Error: 2 UNKNOWN: exit status 1

What's going wrong if someone can explain ????

eloquentarduino, 2021-09-19:
Can you make it work with other boards? Just select "ESP32 Dev Module" for example.

embedragx, 2021-09-20:
In file included from C:\Users\...\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\svdf.cpp:25:
C:\Users\...\Documents\Arduino\libraries\EloquentTinyML-master\src/tensorflow/lite/micro/kernels/activation_utils.h: In function 'float tflite::ops::micro::ActivationValFloat(TfLiteFusedActivation, float)':
C:\Users\...\Documents\Arduino\libraries\EloquentTinyML-master\src/tensorflow/lite/micro/kernels/activation_utils.h:53:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
C:\Users\...\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\svdf.cpp: In function 'TfLiteStatus tflite::ops::micro::svdf::Eval(TfLiteContext*, TfLiteNode*)':
C:\Users\...\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\svdf.cpp:558:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 45 1406 1640 111 (set (reg:SF 19 f0 [orig:198 _248 ] [198])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC101") [flags 0x2]) [0  S4 A32])) "C:\Users\raghu\Documents\Arduino\libraries\EloquentTinyML-master\src/tensorflow/lite/micro/kernels/activation_utils.h":44 47 {movsf_internal}
     (expr_list:REG_EQUAL (const_double:SF 6.0e+0 [0x0.cp+3])
        (nil)))
during RTL pass: postreload
C:\Users\...\Documents\Arduino\libraries\EloquentTinyML-master\src\tensorflow\lite\micro\kernels\svdf.cpp:558:1: internal compiler error: in extract_constrain_insn, at recog.c:2210
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
Using library EloquentTinyML-master at version 0.0.10 in folder: C:\Users\...\Documents\Arduino\libraries\EloquentTinyML-master 
exit status 1
Error compiling for board ESP32 Dev Module.

Same Error ....

eloquentarduino, 2021-09-20:
Try to delete the EloquentTinyML-master folder from the Arduino\libraries folder and install the library via Arduino IDE library manager.

eloquentarduino, 2021-10-31:
I updated the library. Please install from Github since I didn't published on the Arduino Library Manager yet since I made a major update and need to test yet.

arnaudonate, 2021-10-31:
Thank you so much! I will tets it now :) This is very helpful!

embedragx, 2021-11-01:
I will try it soon. 

And thnx for the effort 👍

closed 2024-01-12 by eloquentarduino