eloquentarduino/EloquentTinyML #34: Fail: "cannot move location counter backwards"

iludis21, 2021-11-01:
Hello Simone,
I used your script to create TinyML models, which should recognize 96x96 grayscale images. In Python everything worked as well, model was created. Only compiling for Nano33 BLE in Arduino does not work. It breaks with the error message "linker_script.ld:138 cannot move location counter backwards (from 20fd4a70 to 2003fc00)".
Could you maybe help me further? I get the same error message when I try to compile the exported model from Teachable Machine to Nano 33 BLE.

eloquentarduino, 2021-11-01:
Hi never encountered this error. Please share your Arduino sketch and possibly your Python script.

iludis21, 2021-11-01:
[problem_011121.zip](https://github.com/eloquentarduino/EloquentTinyML/files/7455842/problem_011121.zip)
Here it is. Thanks in advance.

iludis21, 2021-11-03:
[test1_0311.zip](https://github.com/eloquentarduino/EloquentTinyML/files/7469684/test1_0311.zip)
Hello Simone, the problem has been solved. It was beginner's mistake - model was badly made and way too big (28MB). Fixed and optimized to 1,3MB - now it works.
But I have another problem: the input data are int values between 0 and 255, but as output I want to have two float values. But your predict function predict(input, output) supports either int/int or float/float. Can you give me a hint here?

eloquentarduino, 2021-11-03:
I will fix the library on this. For now, declare your input as float so you can use the float/float version of predict.

closed 2024-01-12 by eloquentarduino