eloquentarduino/EloquentTinyML #35: Compilation issues

saisahil2015, 2021-11-27:
The DigitsExample.ino doesn't compile as it shows the following error for the Arduino Nano 33 BLE board.

[image]

Also, if I try compiling it for the Arduino Uno board, then  I get errors as it's not able to recognize my libraries like cstarg, cdtint, etc...!

Please let me know how can resolve the issue for either or both the cases!

eloquentarduino, 2021-11-29:
The examples on the blog are outdated and I'm in the process of rewriting them.
Since EloquentTinyML version 2.4 the API changed. Please read the docs: https://smartmicrotech.it/eloquent-tinyml/getting-started/

If you want to follow the blog, you have to install EloquentTinyML 0.0.10.

Arduino UNO (and any other AVR based board) cannot run TensorFlow.

saisahil2015, 2021-12-02:
Thank you for responding!

When I try compiling the code, I'm getting the following warnings. It's ultimately compiling the code but showing the following warnings. Please let me know as to how we can resolve this and are they quite of a serious issue. Looking at the warnings, they don't seem to be that serious but just want to make sure.

In file included from C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/core/api/op_resolver.h:20:0,
                 from C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/micro/micro_mutable_op_resolver.h:19,
                 from C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/micro/kernels/all_ops_resolver.h:16,
                 from C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/EloquentTinyML.h:13,
                 from C:\Users\saisa\UROP\EloquentTinyML-master\EloquentTinyML-master\examples\DigitsExample\DigitsExample.ino:1:
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h: In function 'const char* tflite::EnumNameTensorType(tflite::TensorType)':
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h:407:9: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (e < TensorType_FLOAT32 || e > TensorType_INT8) return "";
       ~~^~~~~~~~~~~~~~~~~~~~
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h: In function 'const char* tflite::EnumNameQuantizationDetails(tflite::QuantizationDetails)':
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h:437:9: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (e < QuantizationDetails_NONE || e > QuantizationDetails_CustomQuantization) return "";
       ~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/core/api/op_resolver.h:20:0,
                 from C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/micro/micro_mutable_op_resolver.h:19,
                 from C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/micro/kernels/all_ops_resolver.h:16,
                 from C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/EloquentTinyML.h:13,
                 from C:\Users\saisa\UROP\EloquentTinyML-master\EloquentTinyML-master\examples\DigitsExample\DigitsExample.ino:1:
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h: In function 'const char* tflite::EnumNameDimensionType(tflite::DimensionType)':
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h:520:9: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (e < DimensionType_DENSE || e > DimensionType_SPARSE_CSR) return "";
       ~~^~~~~~~~~~~~~~~~~~~~~
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h: In function 'const char* tflite::EnumNameBuiltinOperator(tflite::BuiltinOperator)':
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h:922:9: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (e < BuiltinOperator_ADD || e > BuiltinOperator_SEGMENT_SUM) return "";
       ~~^~~~~~~~~~~~~~~~~~~~~
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h: In function 'const char* tflite::EnumNameBuiltinOptions(tflite::BuiltinOptions)':
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h:1249:9: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (e < BuiltinOptions_NONE || e > BuiltinOptions_SegmentSumOptions) return "";
       ~~^~~~~~~~~~~~~~~~~~~~~
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h: In function 'const char* tflite::EnumNamePadding(tflite::Padding)':
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h:2520:9: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (e < Padding_SAME || e > Padding_VALID) return "";
       ~~^~~~~~~~~~~~~~
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h: In function 'const char* tflite::EnumNameActivationFunctionType(tflite::ActivationFunctionType)':
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h:2562:9: warning: comparison is always false due to limited range of data type [-Wtype-limits]
   if (e < ActivationFunctionType_NONE || e > ActivationFunctionType_SIGN_BIT) return "";
       ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/schema/schema_generated.h: In function 'const char* tflite::EnumNameLSHProjectionType(tflite::LSHProjectionType)':
[... cut ...]
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/micro/kernels/activation_utils.h: In function 'float tflite::ops::micro::ActivationValFloat(TfLiteFusedActivation, float)':
C:\Users\saisa\OneDrive\Documents\Arduino\libraries\EloquentTinyML\src/tensorflow/lite/micro/kernels/activation_utils.h:53:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Sketch uses 208608 bytes (21%) of program storage space. Maximum is 983040 bytes.
Global variables use 57920 bytes (22%) of dynamic memory, leaving 204224 bytes for local variables. Maximum is 262144 bytes.

eloquentarduino, 2021-12-02:
I can see those warnings are from the TensorFlow library itself, I'm sorry I don't have any control over it.
It looks like you turned on restrictive warnings and they're complaining about bad comparisons, which I'm pretty sure it intended behavior from the TensorFlow team.

To be sure, you should write on their repo.

As a sidenote, I see you're using an outdated version of the EloquentTinyML library, I suggest you upgrade: https://smartmicrotech.it/eloquent-tinyml/getting-started/

saisahil2015, 2021-12-04:
I was using 2.4.0 but as I've mentioned above, there are some issues that I was facing. Then, you share a blog and recommended to use the version 0.0.10 for it and so I did and received the above warnings. 

I know that you said that you are currently updating the blog for the version 2.4.0. Please let me know as to when it will be updated by! Thank you again for your response and help!

closed 2022-09-02 by eloquentarduino