eloquentarduino/EloquentTinyML #37: Error compiling for  ESP32-DEV-Module

nicoro-yja, 2021-12-31:
Hi，The following error is reported at compile time：

c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_1x1_HWC_q7_fast_nonsquare.c:1:0: fatal error: opening dependency file C:\Users\HP\AppData\Local\Temp\arduino-sketch-4FBA319415F0427AB022FB1B916D6EC9\libraries\EloquentTinyML\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_1x1_HWC_q7_fast_nonsquare.c.d: No such file or directory
 #if !defined(ESP32)
 ^compilation terminated.
Compilation error: Error: 13 INTERNAL: exit status 1

help me please
thanks

eloquentarduino, 2021-12-31:
Which board are you compiling for?

nicoro-yja, 2021-12-31:
Thank you for your prompt reply，My board is ESP32-WROOM-32E

nicoro-yja, 2021-12-31:
IDE is Arduino 2.0

eloquentarduino, 2021-12-31:
Do you have an Arduino IDE 1 installation to try? If yes, please report if it compiles.

eloquentarduino, 2021-12-31:
It is working fine for me on Arduino IDE 2.0.
I cannot find the ESP32-WROOM-32E board in my IDE, can you please share the board definition json?

nicoro-yja, 2021-12-31:
Ok, I'll try it on right now
Thank you。

nicoro-yja, 2021-12-31:
I add it，it report error like this：
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_HWC_q15_fast_nonsquare.c:1:0: fatal error: opening dependency file C:\Users\HP\AppData\Local\Temp\arduino-sketch-4FBA319415F0427AB022FB1B916D6EC9\libraries\EloquentTinyML\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_HWC_q15_fast_nonsquare.c.d: No such file or directory
 #if !defined(ESP32)
 ^
compilation terminated.
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_1x1_HWC_q7_fast_nonsquare.c:1:0: fatal error: opening dependency file C:\Users\HP\AppData\Local\Temp\arduino-sketch-4FBA319415F0427AB022FB1B916D6EC9\libraries\EloquentTinyML\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_1x1_HWC_q7_fast_nonsquare.c.d: No such file or directory
 #if !defined(ESP32)
 ^
compilation terminated.
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_HWC_q7_basic_nonsquare.c:1:0: fatal error: opening dependency file C:\Users\HP\AppData\Local\Temp\arduino-sketch-4FBA319415F0427AB022FB1B916D6EC9\libraries\EloquentTinyML\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_HWC_q7_basic_nonsquare.c.d: No such file or directory
 #if !defined(ESP32)
 ^
compilation terminated.
Compilation error: Error: 13 INTERNAL: exit status 1

eloquentarduino, 2021-12-31:
What sketch are you trying to compile?

nicoro-yja, 2022-01-01:
SineExample.ino

nicoro-yja, 2022-01-01:
Happy New Year!

I here is xx this example, in the environment of Win10, will report that errors.
help me ，Please.
Thanks^_^

nicoro-yja, 2022-01-02:
I'm on another computer, Windows 10, Arduino1.8，and it still reports errors

eloquentarduino, 2022-01-02:
I will take a look tomorrow on Windows

nicoro-yja, 2022-01-03:
OK,Thanks.^-^

eloquentarduino, 2022-01-04:
I can confirm it works on my Windows 10 Arduino 1.8.10, with both targets Arduino RP2040 Connect and ESP32 Dev Module.
You may try to delete the library manually from your PC and re-installing version 2.4.0-bis (it's almost identical to 2.4.0 really).

nicoro-yja, 2022-01-05:
Ok, thank you. I'll try another computer

eloquentarduino, 2022-01-05:
Try to compile for an ARM board and see if that works at least.

nicoro-yja, 2022-01-06:
Hello, I use the model Arduino MKRZERO. It's ARM, and it still reports an error

nicoro-yja, 2022-01-06:
__STATIC_FORCEINLINE uint32_t __SMLADX(
   ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/none.h:484:3: error: '__STATIC_FORCEINLINE' does not name a type
   __STATIC_FORCEINLINE uint32_t __SMLSDX(
   ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/none.h:498:3: error: '__STATIC_FORCEINLINE' does not name a type
   __STATIC_FORCEINLINE uint64_t __SMLALD(
   ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/utils.h:156:1: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'?
 __STATIC_INLINE  void arm_norm_64_to_32u(uint64_t in, int32_t * normalized, int32_t *norm)
 ^~~~~~~~~~~~~~~
 _ELIDABLE_INLINE
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/utils.h:208:1: error: '__STATIC_INLINE' does not name a type; did you mean '_ELIDABLE_INLINE'?
 __STATIC_INLINE q31_t arm_div_q63_to_q31(q63_t num, q31_t den)
 ^~~~~~~~~~~~~~~
 _ELIDABLE_INLINE
In file included from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/statistics_functions.h:37:0,
                 from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/bayes_functions.h:36,
                 from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:210,
                 from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:34,
                 from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:165,
                 from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\kernels\cmsis-nn\fully_connected.cpp:19:
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/fast_math_functions.h:197:1: error: '__STATIC_FORCEINLINE' does not name a type
 __STATIC_FORCEINLINE arm_status arm_sqrt_f32(
 ^~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:214:0,
                 from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:34,
                 from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:165,
                 from c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\kernels\cmsis-nn\fully_connected.cpp:19:
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/controller_functions.h:269:3: error: '__STATIC_FORCEINLINE' does not name a type
   __STATIC_FORCEINLINE float32_t arm_pid_f32(
   ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/none.h:331:3: error: '__STATIC_FORCEINLINE' does not name a type
   __STATIC_FORCEINLINE uint32_t __SHSUB16(
   ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/none.h:347:3: error: '__STATIC_FORCEINLINE' does not name a type
   __STATIC_FORCEINLINE uint32_t __QASX(
   ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/none.h:363:3: error: '__STATIC_FORCEINLINE' does not name a type
   __STATIC_FORCEINLINE uint32_t __SHASX(
   ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/none.h:379:3: error: '__STATIC_FORCEINLINE' does not name a type
   __STATIC_FORCEINLINE uint32_t __QSAX(
   ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/dsp/none.h:395:3: error: '__STATIC_FORCEINLINE' does not name a type
   __STATIC_FORCEINLINE uint32_t __SHSAX(
   ^~~~~~~~~~~~~~~~~~~~
[... cut ...]
 __STATIC_FORCEINLINE q31_t arm_nn_mult_by_power_of_two(const int32_t val, const int32_t exp)
 ^~~~~~~~~~~~~~~~~~~~
c:\Users\HP\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:937:1: error: '__STATIC_FORCEINLINE' does not name a type
 __STATIC_FORCEINLINE int32_t arm_nn_one_over_one_plus_x_for_x_in_0_1(int32_t val)
 ^~~~~~~~~~~~~~~~~~~~
Compilation error: Error: 13 INTERNAL: exit status 1

eloquentarduino, 2022-01-07:
I never encountered these kind of errors. I looks like something is broken with the CMSIS framework.
Please install the Arduino_TensorFlowLite library from the library manager and try to compile the "hello world" example: does it compile fine?

nicoro-yja, 2022-01-07:
I installed TensorFlowLite_ESP32.h and compiled the "hello world" example，it‘s OK.

eloquentarduino, 2022-01-11:
I'm really lost here, I cannot find anything online about your error. I'm sure this is about the CMSIS framework, but it shouldn't appear when compiling against the ESP32. Please, try to delete the EloquentTinyML folder and clone from this repo instead of installing via the Library Manager. Then go to `src/eloquent_tinyml/tensorflow.h` and delete the following

```cpp
#define ELOQUENT_TINYML_ARM 1
#include "./tensorflow/arm/tensorflow/lite/version.h"
#include "./tensorflow/arm/tensorflow/lite/schema/schema_generated.h"
#include "./tensorflow/arm/tensorflow/lite/micro/micro_error_reporter.h"
#include "./tensorflow/arm/tensorflow/lite/micro/all_ops_resolver.h"
#include "./tensorflow/arm/tensorflow/lite/micro/micro_mutable_op_resolver.h"
#include "./tensorflow/arm/tensorflow/lite/micro/micro_interpreter.h"
#include "./tensorflow/arm/patches/AllOpsResolver.h"
#include "./tensorflow/arm/patches/MicroMutableOpResolver.h"
```

The try to compile for both ESP32 and ARM.

nicoro-yja, 2022-01-13:
Thank you, I will try again, maybe I have a problem with the usage, several different machines will report this error。
So what's the difference between Tensorflow-ESP and EloquentTinyML?
Thanks

eloquentarduino, 2022-01-13:
Tensorflow-ESP ships a pre-compiled version of TensorFlow targeted at the ESP32.
EloquentTinyML copies over the files from TensorFlow-ESP32 and adds an abstraction layer on top of it to: 1) make it consistent with the ARM implementation (a few classes have different APIs) and 2) make it easier to use (avoid writing boilerplate code)

pixelEDI, 2022-02-13:
Unfortunately I got the same error, also with the "#define ESP32 1" 

`Source\ConvolutionFunctions\arm_convolve_HWC_q15_fast_nonsquare.c.d: No such file or directory
 #if !defined(ESP32)
 ^
compilation terminated.
`

Arduino IDE 2.0 / ESP32 / EloquentTinyML 2.4.1/2

eloquentarduino, 2022-02-14:
1) File should be `arm_convolve_HWC_q15_fast_nonsquare.c`, without the `.d` at the end. Which file is raising this error?
2) There is something not working as it should, since it is still trying to load the ARM CMSIS framework, not the ESP32 one.

The extreme solution I can suggest is:

 - delete the `eloquent_tinyml/tensorflow/arm` folder altogether
 - delete

```cpp
#define ELOQUENT_TINYML_ARM 1
    #include "./tensorflow/arm/tensorflow/lite/version.h"
    #include "./tensorflow/arm/tensorflow/lite/schema/schema_generated.h"
    #include "./tensorflow/arm/tensorflow/lite/micro/micro_error_reporter.h"
    #include "./tensorflow/arm/tensorflow/lite/micro/all_ops_resolver.h"
    #include "./tensorflow/arm/tensorflow/lite/micro/micro_mutable_op_resolver.h"
    #include "./tensorflow/arm/tensorflow/lite/micro/micro_interpreter.h"
    #include "./tensorflow/arm/patches/AllOpsResolver.h"
    #include "./tensorflow/arm/patches/MicroMutableOpResolver.h"
```

from `eloquent_tinyml/tensorflow.h`

 - delete the `#if defined(ESP32)`, keeping all the esp32-related includes.

In practive, you have to remove all references to ARM and only keep references to ESP32 files.

pixelEDI, 2022-02-14:
Great - that worked great for me! Thanks a lot!

eloquentarduino, 2022-02-15:
Do you feel like it would be less troublesome to "force" the user to include the correct version of TensorFlow?
Instead of

```cpp
#include <eloquent_tinyml/tensorflow.h>
```

```cpp
#include <eloquent_tinyml/tensorlow/esp32.h>
or
#include <eloquent_tinyml/tensorflow/arm.h>
```

pixelEDI, 2022-02-15:
I think this is not necessary, because the error somehow only occurred with me and someone else.

Keep up you great work!

tilman, 2023-01-20:
I had the same error. I think making it choosable would be better.

tebrizli, 2023-05-12:
> 1. File should be `arm_convolve_HWC_q15_fast_nonsquare.c`, without the `.d` at the end. Which file is raising this error?
> 2. There is something not working as it should, since it is still trying to load the ARM CMSIS framework, not the ESP32 one.
> 
> The extreme solution I can suggest is:
> 
> * delete the `eloquent_tinyml/tensorflow/arm` folder altogether
> * delete
> 
> ```c++
> #define ELOQUENT_TINYML_ARM 1
>     #include "./tensorflow/arm/tensorflow/lite/version.h"
>     #include "./tensorflow/arm/tensorflow/lite/schema/schema_generated.h"
>     #include "./tensorflow/arm/tensorflow/lite/micro/micro_error_reporter.h"
>     #include "./tensorflow/arm/tensorflow/lite/micro/all_ops_resolver.h"
>     #include "./tensorflow/arm/tensorflow/lite/micro/micro_mutable_op_resolver.h"
>     #include "./tensorflow/arm/tensorflow/lite/micro/micro_interpreter.h"
>     #include "./tensorflow/arm/patches/AllOpsResolver.h"
>     #include "./tensorflow/arm/patches/MicroMutableOpResolver.h"
> ```
> 
> from `eloquent_tinyml/tensorflow.h`
> 
> * delete the `#if defined(ESP32) ... #else ... #endif` wrapper, keeping all the esp32-related includes.
> 
> In practive, you have to remove all references to ARM and only keep references to ESP32 files.

Great!!!!!!!!!!!!  Thats work!

eloquentarduino, 2023-05-12:
I'm leaning to split the libraries into EloquentTinyMLArm and EloquentTinyMLEsp32, because the unification of both is causing troubles to a lot of people.

eloquentarduino, 2023-12-07:
In the end I split the libraries. The ESP32 version is at https://github.com/eloquentarduino/EloquentTensorFlow32

closed 2023-12-07 by eloquentarduino