eloquentarduino/EloquentTinyML #38: Issue on compiling Examples>2.4>SineExample on Arduino Nano 33 BLE Sense board

sanjevShakya, 2022-01-21:
Arduino version: 1.8.19
Arduino nano 33 BLE board version: 2.6.1
 
```Arduino: 1.8.19 (Linux), Board: "Arduino Nano 33 BLE"

Invalid library found in /home/sanjeev/Arduino/libraries/arm_math: no headers files (.h) found in /home/sanjeev/Arduino/libraries/arm_math
In file included from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:34:0,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:165,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/kernels/cmsis-nn/svdf.cpp:21:
/home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:199:44: error: 'uint32_t __SXTB16' redeclared as different kind of symbol
 #define __SXTB16_RORn(ARG1, ARG2) __SXTB16(__ROR(ARG1, ARG2))
                                            ^
/home/sanjeev/.arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_gcc.h:1989:31: note: in expansion of macro '__SXTB16_RORn'
 __STATIC_FORCEINLINE uint32_t __SXTB16_RORn(uint32_t op1, uint32_t rotate)
                               ^~~~~~~~~~~~~
In file included from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/Core/Include/cmsis_compiler.h:55:0,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math_types.h:77,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:202,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:34,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:165,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/kernels/cmsis-nn/svdf.cpp:21:
/home/sanjeev/.arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_gcc.h:1981:31: note: previous declaration 'uint32_t __SXTB16(uint32_t)'
 __STATIC_FORCEINLINE uint32_t __SXTB16(uint32_t op1)
                               ^~~~~~~~
In file included from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:34:0,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:165,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/kernels/cmsis-nn/svdf.cpp:21:
/home/sanjeev/.arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_gcc.h:1989:54: error: expected primary-expression before 'op1'
 __STATIC_FORCEINLINE uint32_t __SXTB16_RORn(uint32_t op1, uint32_t rotate)
                                                      ^
/home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:199:50: note: in definition of macro '__SXTB16_RORn'
 #define __SXTB16_RORn(ARG1, ARG2) __SXTB16(__ROR(ARG1, ARG2))
                                                  ^~~~
/home/sanjeev/.arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_gcc.h:1989:68: error: expected primary-expression before 'rotate'
 __STATIC_FORCEINLINE uint32_t __SXTB16_RORn(uint32_t op1, uint32_t rotate)
                                                                    ^
/home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:199:56: note: in definition of macro '__SXTB16_RORn'
 #define __SXTB16_RORn(ARG1, ARG2) __SXTB16(__ROR(ARG1, ARG2))
                                                        ^~~~
In file included from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:34:0,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:165,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/kernels/cmsis-nn/fully_connected.cpp:19:
/home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:199:44: error: 'uint32_t __SXTB16' redeclared as different kind of symbol
 #define __SXTB16_RORn(ARG1, ARG2) __SXTB16(__ROR(ARG1, ARG2))
                                            ^
/home/sanjeev/.arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_gcc.h:1989:31: note: in expansion of macro '__SXTB16_RORn'
 __STATIC_FORCEINLINE uint32_t __SXTB16_RORn(uint32_t op1, uint32_t rotate)
                               ^~~~~~~~~~~~~
In file included from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/Core/Include/cmsis_compiler.h:55:0,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math_types.h:77,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:202,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:34,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:165,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/kernels/cmsis-nn/fully_connected.cpp:19:
/home/sanjeev/.arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_gcc.h:1981:31: note: previous declaration 'uint32_t __SXTB16(uint32_t)'
 __STATIC_FORCEINLINE uint32_t __SXTB16(uint32_t op1)
                               ^~~~~~~~
In file included from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnsupportfunctions.h:34:0,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Include/arm_nnfunctions.h:165,
                 from /home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/kernels/cmsis-nn/fully_connected.cpp:19:
/home/sanjeev/.arduino15/packages/arduino/hardware/mbed_nano/2.6.1/cores/arduino/mbed/cmsis/CMSIS_5/CMSIS/TARGET_CORTEX_M/Include/cmsis_gcc.h:1989:54: error: expected primary-expression before 'op1'
 __STATIC_FORCEINLINE uint32_t __SXTB16_RORn(uint32_t op1, uint32_t rotate)
                                                      ^
[... cut ...]
 __STATIC_FORCEINLINE uint32_t __SXTB16_RORn(uint32_t op1, uint32_t rotate)
                                                                    ^
/home/sanjeev/Arduino/libraries/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/DSP/Include/arm_math.h:199:56: note: in definition of macro '__SXTB16_RORn'
 #define __SXTB16_RORn(ARG1, ARG2) __SXTB16(__ROR(ARG1, ARG2))
                                                        ^~~~
exit status 1
Error compiling for board Arduino Nano 33 BLE.

```

sanjevShakya, 2022-01-22:
Library version used: 2.4.1 (downloaded as a zip file and added to the libs)
Same error when using platformIO with board arduino ble sense 

[image]

sanjevShakya, 2022-01-22:
I commented out this particular line of code 
```
// #define __SXTB16_RORn(ARG1, ARG2) __SXTB16(__ROR(ARG1, ARG2))
```
and I was able to compile and run the example. I am not really sure what this is doing can you help me know what it is doing?

eloquentarduino, 2022-01-23:
I'm aware of this error.
I need to setup an extensive test suite for each board and patch this kind of errors for each one.
For the moment, keep that line commented and test if everything works fine, I guess this has something to do with the CMSIS or Mbed OS definitions for each board.

sanjevShakya, 2022-01-23:
Thank you for the reply, yes I was able to deploy a custom CNN model and inference from it.

closed 2024-01-12 by eloquentarduino