eloquentarduino/EloquentTinyML #41: Issue on the Colab

chisu7917, 2022-02-14:
Hi Simone,

Thanks for your very nice code! Really appreciate it!
I have the same issue with Radhanand Anantha who post a question on your blog.  [http://disq.us/p/2e6vmdl](url)
I also use colab to generate the model and finally the model size is about 5000 instead of 21448. The code is totally the same with yours because I just copied it to verify. 
Here is the link of colab code. Could you please have a look? Thanks!
[https://colab.research.google.com/drive/1_KE3OJ5uWD4bD-DITsQ4N6J8N8KDRorT?usp=sharing](url)

eloquentarduino, 2022-02-14:
Does the model of size 5000 works fine or it is not able to classify digits?

chisu7917, 2022-02-14:
Hi Simone,

No, it doesn't work. it looks like as below. 

18:40:37.835 -> It took 3 micros to run inference
18:40:37.835 -> Test output is: 8
18:40:37.835 -> Predicted proba are: 0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00
18:40:37.835 -> Predicted class is: 0
18:40:37.835 -> Sanity check: 7

eloquentarduino, 2022-02-14:
You probably have an outdated version of tinymlgen. Either

 1. upgrade tinymlgen
 2. use `port(model, variable_name='digits_model', pretty_print=True, optimize=False)`

The latest version of tinymlgen makes `optimize=False` the default, prior versions had `optimize=True`.
It is now working fine for me.

chisu7917, 2022-02-14:
Thanks for your prompt reply, Simone!

Yes, you are right. I have modified the code and got the right results of classification!! I tried both methods your mentioned and got the same result.

I'm still curious why the length of my model is "digits_model_len = 13504;" instead of 21448? Both your methods generated the same length of model of 13504 on my side.

Thanks!

https://colab.research.google.com/drive/1_KE3OJ5uWD4bD-DITsQ4N6J8N8KDRorT?usp=sharing

eloquentarduino, 2022-02-15:
Maybe that 21448 was for a different model, who knows...

closed 2022-09-02 by eloquentarduino