eloquentarduino/EloquentTinyML #43: 'Didn't find op for builtin opcode ...' error

expeon07, 2022-03-24:
Hi! I'm trying to upload a TFLite model that uses autoencoder and LSTM but I get the following errors when uploading

(for Autoencoder)
```
Start
GetModel done
Version check done
Didn't find op for builtin opcode 'TANH' version '1'

Failed to get registration from op code  d
 
Allocation failed
AllocateTensors() failed
```

(for LSTM)
```
Start
GetModel done
Version check done
Didn't find op for builtin opcode 'FULLY_CONNECTED' version '9'

Failed to get registration from op code  d
 
Allocation failed
AllocateTensors() failed
```

Is there any special configuration that I should do or are these not supported yet?

eloquentarduino, 2022-03-25:
Tanh seems to be supported: https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/all_ops_resolver.cc
Do you have any chance to replace tanh with something else to test if the rest of network works?

expeon07, 2022-03-25:
Yes, I substituted it with sigmoid and relu and they didn't get an error.

eloquentarduino, 2022-03-25:
As far as I know, to implement RNN you should use GRU instead of LSTM.
GRU should be supported.

expeon07, 2022-03-26:
I tried substituting the LSTM layer with the GRU as well, but it gave the same error.
I also closely followed this implementation here: 
https://colab.research.google.com/github/tensorflow/tensorflow/blob/master/tensorflow/lite/examples/experimental_new_converter/Keras_LSTM_fusion_Codelab.ipynb#scrollTo=tB1NZBUHDogR
but to no avail. :(

eloquentarduino, 2022-03-27:
I will take a look at LSTM/GRU this week and see if I find something that
works. Be patient.

expeon07, 2022-03-27:
Thanks a lot! That will be very helpful.

eloquentarduino, 2022-09-02:
LSTM is indeed supported as many users reported.
Be sure to use Tf==2.2.3

closed 2022-09-02 by eloquentarduino