eloquentarduino/EloquentTinyML #44: Support for RNN (LSTM)

expeon07, 2022-03-25:
Hi, would you know if there is a way to implement RNN in TFLite micro? I use LSTM for time series prediction but I have this error. 

`Only 1 subgraph is currently supported.`

dustinjoe, 2022-04-16:
I actually had the same question when I tried to run a LSTM model last summer for a paper submission:
https://github.com/dustinjoe/TinyML-with-Cloud-for-PHM
At that time, no official support for RNNs, I remember someone from Tensorflow team mentioned some memory issue.
But this time point, official LSTM support seems to be on the way right now:
https://github.com/tensorflow/tflite-micro/pull/1044

eloquentarduino, 2022-09-02:
LSTM is indeed supported as many users reported. Be sure to use Tf==2.2.3

closed 2022-09-02 by eloquentarduino

whubaichuan, 2022-11-04:
@expeon07 @dustinjoe @eloquentarduino There is an operation called UnidirectionalSequenceLSTM in [the official library description](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/all_ops_resolver.cc). But I am not sure whether that's LSTM you want or not.