eloquentarduino/EloquentTinyML #45: 'Eloquent' does not name a type

hanyc93, 2022-04-19:
I install the library via: Arduino IDE > Tools > Manage Libraries
Board: ESP32 Dev Module

Error when compile:

SineExample:11:1: error: 'Eloquent' does not name a type
 Eloquent::TinyML::TfLite<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml;
 ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_modified_sketch_199840\SineExample.ino: In function 'void setup()':
SineExample:16:5: error: 'ml' was not declared in this scope
     ml.begin(sine_model);
     ^
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_modified_sketch_199840\SineExample.ino: In function 'void loop()':
SineExample:25:27: error: 'ml' was not declared in this scope
         float predicted = ml.predict(input);
                           ^
exit status 1
'Eloquent' does not name a type

SAOKIM99, 2022-04-27:
try using version 0.5.
I also got this error and reverted to an older version to be able to run

eloquentarduino, 2022-09-02:
This is due to a breaking change in the API.
Install version 2.4.3 then

```cpp
#include <EloquentTinyML.h>
#include <eloquent_tinyml/tensorflow.h>

Eloquent::TinyML::TensorFlow::TensorFlow<N_INPUTS, N_OUTPUTS, TENSOR_ARENA_SIZE> tf;
```

closed 2022-09-02 by eloquentarduino

YangSongL1n, 2022-12-14:
Solved  cool!

YangSongL1n, 2022-12-14:
> This is due to a breaking change in the API. Install version 2.4.3 then
> 
> ```c++
> #include <EloquentTinyML.h>
> #include <eloquent_tinyml/tensorflow.h>
> 
> Eloquent::TinyML::TensorFlow::TensorFlow<N_INPUTS, N_OUTPUTS, TENSOR_ARENA_SIZE> tf;
> ```
Solved cool!

nhandt64, 2025-03-27:
> This is due to a breaking change in the API. Install version 2.4.3 then
> 
> #include <EloquentTinyML.h>
> #include <eloquent_tinyml/tensorflow.h>
> 
> Eloquent::TinyML::TensorFlow::TensorFlow<N_INPUTS, N_OUTPUTS, TENSOR_ARENA_SIZE> tf;

A little hello from future here, which is 2 and a half year later.

I found you enjoyed rewriting the API so much that I can't either use this old library (with tremendous amount of mismatch variable type to missing library) or new library just because I have to explicitly declare the size of input and output.

I reverted to version 2.4.3, literally copied the above code and basically it just yells at me

`
Compilation error: 'Eloquent' does not name a type
`

Even on the latest version, the docs are far from able to figure out how to use it and constant coming out from nowhere, for instance:
`/**
 * Run a TensorFlow model to predict the IRIS dataset
 * For a complete guide, visit
 * https://eloquentarduino.com/tensorflow-lite-esp32
 */
// replace with your own model
// include BEFORE <eloquent_tinyml.h>!
#include "irisModel.h"
// include the runtime specific for your board
// either tflm_esp32 or tflm_cortexm
#include <tflm_esp32.h>
// now you can include the eloquent tinyml wrapper
#include <eloquent_tinyml.h>

// this is trial-and-error process
// when developing a new model, start with a high value
// (e.g. 10000), then decrease until the model stops
// working as expected
#define ARENA_SIZE 2000

Eloquent::TF::Sequential<TF_NUM_OPS, ARENA_SIZE> tf;
`
where on Earth does that `TF_NUM_OPS` come from?