eloquentarduino/EloquentTinyML #46: ...arm_convolve_1x1_HWC_q7_fast_nonsquare.c.d: No such file or directory

ts532, 2022-08-25:
Using ESP-32 Feather Board in Vscode with PlatformIO.
Full Error:
lib/EloquentTinyML/src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Source/ConvolutionFunctions/arm_convolve_1x1_HWC_q7_fast_nonsquare.c:1: fatal error: opening dependency file .pio/build/featheresp32/lib93d/EloquentTinyML/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/tools/make/downloads/cmsis/CMSIS/NN/Source/ConvolutionFunctions/arm_convolve_1x1_HWC_q7_fast_nonsquare.c.d: No such file or directory

It seems that PlatformIO isn't finding/making a "arm_convolve_1x1_HWC_q7_fast_nonsquare.c.d" file within .pio. Although when checking the EloquentTinyML library an "arm_convolve_1x1_HWC_q7_fast_nonsquare.c" does seem to exist where it should, i'm not sure if these are related.

I am using the latest version, v2.4.0, although I manually added this library instead of via PlatformIO as it seems to only have v0.02 and v0.03 available.

I have been unable to make this library work with either Vscode or Arduino IDE (with Arduino IDE, a similar "No Such File or Directory" Error occurs, although this time for "C:\Users\[Name]\Documents\Arduino\libraries\EloquentTinyML\ src/eloquent_tinyml/tensorflow/arm/third_party/flatbuffers/include/flatbuffers/base.h:32:10: fatal error: cstddef: No such file or directory"

Is there any advice to resolve these errors? Making recognition systems play nice with microcontrollers has been a brick wall.

ts532, 2022-08-28:
I believe I have found the source of the issue. The file path lengths were so long they surpassed the 260 Character limit imposed by Windows, preventing files from being generated by PlatformIO. Oddly, disabling file path max lengths in the Registry Editor had no effect, and still caused the same errors. 

Moving the directory folder I stored my code in to be directly in C:/ (so less subdirectories and shorter file path) resolved the issue, and the program built normally.

eloquentarduino, 2022-09-02:
Glad you solved.

closed 2022-09-02 by eloquentarduino

whubaichuan, 2022-10-19:
@ts532 hi, thanks for your answer. You talk about changing the code path. Do you mean the path of the ARDUINO project or the path of EloquentTinyML library code?

ts532, 2022-10-19:
> @ts532 hi, thanks for your answer. You talk about changing the code path. Do you mean the path of the ARDUINO project or the path of EloquentTinyML library code?

It would be a bit of both. I use VSCode with PlatformIO to allow me to write Arduino Code. PlatformIO generates files I assume are used during the build process in a .pio folder within my project folder. These can end up with very long path lengths as they are copies of the EloquentTinyML Library. The total file path length would consist of the project location (in italics) and the library path therefore looks something like this:

_C:\Users\name\Documents\GitHub\PhDStuff\Software\ESPArduino\_._pio\build\featheresp32\lib001\EloquentTinyML\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\tools\make\downloads\cmsis\CMSIS\NN\Source\ConvolutionFunctions\arm_convolve_HWC_q7_fast_nonsquare.c.d_

As I can't control the length of the library path, I moved the project to a different location so that the project file path was shorter (less folders to go through), and so wouldn't run into the path limit. After I did this, PlatformIO was able to find all files and build successfully despite no changes being made to the code I was running.

eloquentarduino, 2022-10-20:
@ts532 I will add a warning on the README suggesting to create the project near the root of C:\ on Windows. It seems hard to debug otherwise.

Juatin8, 2023-12-07:
I encountered the same problem as well. Seems I have to change the location of Arduino library as well

eloquentarduino, 2023-12-07:
For ESP32, I separated the source code into a new library: https://github.com/eloquentarduino/EloquentTensorFlow32
Please refer to this new library for future developments.

whubaichuan, 2023-12-07:
@Juatin8 You can try to use MacOS or Online Arduino Editor to avoid the path problem.