eloquentarduino/EloquentTinyML #50: Problem in apply Conv2D on my own case when refer to your digits example.

whubaichuan, 2022-11-04:
Hi, recently I am doing an interesting project about using Conv2D or Dense to detect 2-classification-issues. I refer to [your digits example](https://github.com/eloquentarduino/EloquentTinyML/tree/master/examples/DigitsExample).

I see in your digits example, you train the model using `N*8*8*1` data. But in [DigitsExample.ino](https://github.com/eloquentarduino/EloquentTinyML/blob/master/examples/DigitsExample/DigitsExample.ino), you input the `1*64` data to test. If it was in the Keras environment, the test input should also be `1*8*8*1`.  Does that mean in Arduino environment, the EloquentTinyML will regards all the input in one dimension structure as the `1*8*8*1` structure according to the `.h`  network file, as you said [here](https://github.com/eloquentarduino/EloquentTinyML/issues/28).

Next, it's my own project. Actually, my training data is `N*1024`. Next, I tested two different types of models. One is Conv2D and Dense, and other is fully Dense. I am sure I have made `optimize=False` as told [here](https://github.com/eloquentarduino/EloquentTinyML/issues/15).

1.  **Conv2D and Dense** _doesn't work_

where the input is `1*32*32*1` and the output is 2-classification.  I refer to your digits example, have the model in Keras like this:

```
model = tf.keras.Sequential()
model.add(layers.Conv2D(8, (3, 3), activation='relu', input_shape=(32, 32, 1)))
model.add(layers.Flatten()) 
#Flatten works in your digits example although Flatten seems to be unsupport by TFLite. 
#But anyway, I also have tried layers.Reshape() here, which should be supported by TFLite.
model.add(layers.Dense(2))
```

In Keras training, I reshape my training data into `N*32*32*1`. In the Arduino project, I reshape my data into `1*1024`, which is similar to your `1*64` in digits. But the `Predicted proba are 0.0 , 0.0`, which is so weird.

2. **fully Dense** _works_

where the input is `1*1024` and the output is 2-classification.  The model is:

```
model = tf.keras.models.Sequential([
tf.keras.layers.Dense(10, activation=tf.nn.relu,input_shape=(1024,)),
tf.keras.layers.Dense(2)
])
```

In Keras training, I reshape my training data into `N*1024`. In the Arduino project, I also reshape my data into `1*1024`, which is similar to your `1*64` in digits. It works normally. The `Predicted proba are 0.90 , 0.10`, which is ok.

Now I am stuck with this tricky problem. I guess the reason is the data structure problem. But perhaps there are another aspects I haven't considered. Do you have another advice for me?

Looking forward to your reply.

Best wishes

eloquentarduino, 2022-11-05:
Can I see the data set?
Also, which version of TF are you using?

whubaichuan, 2022-11-05:
@eloquentarduino hi, one example of data is:
```
float x_test[1024] = {-0.06448305058721761,-0.17836999989388502,-0.12258710468552886,-0.31052685648476974,-0.38262579301428384,-0.3332177841866167,-0.40394477057068706,-0.1661914527554845,-0.14370265725357373,-0.16496679454972513,-0.19555417756958987,-0.17729064362243888,-0.09682711914316249,-0.12436193145423562,-0.1552695051283168,-0.20016971341821915,-0.21710855284598662,-0.16469292917817305,-0.07472331069344791,0.013321807424841136,-0.05865138405551554,-0.05908285192701577,-0.03838014135621375,-0.023760041297648818,0.06254668115531059,0.11348508492733819,0.08935422259379233,0.17486494234706296,-0.056143649125266364,-0.08200761398609457,-0.032743941753975936,0.20970614111909164,0.24237065999625093,0.3797523925510097,0.43123676165710917,0.24952161632963307,0.23160244651857426,0.27643685696234094,0.18398822146844238,0.16058144933259016,0.24321203957682766,0.0020172972602788084,0.0690625509579853,0.0870467084454029,0.15074376815432308,0.21352603972460693,0.24342240874300744,0.3228789375538727,0.288208804445636,0.2740316126048805,0.13452704828999323,0.08057527065710975,0.03263920894519019,0.04198722546288908,0.07067523880220428,0.12644597589235243,0.0833620921810991,0.008706925106669608,0.10125722766354792,0.11689479280159926,0.07089623228143606,0.25878507572845855,0.11046063520557435,0.07401894733855446,0.144557481899811,0.22638362010072777,0.23184348010761385,0.22513240666121534,0.2062193516795875,0.03835969616626138,0.15133154641147947,0.22773095451982722,0.19064393166170746,0.38047114165082707,0.24303932138960563,0.0917277213094672,0.25732968453458954,0.2300157496403091,0.23257037618654924,0.5060929550303774,0.3547744841323576,0.019807197115212064,-0.017605048137457793,-0.21963597436819662,-0.20568877189072382,0.06517211320190651,0.030858409781287063,0.0017171751368199857,-0.14446776263996097,-0.19552776371000685,-0.17225271217942226,-0.19845738340150862,-0.26217588677556325,-0.2647565268944236,-0.14072833565582307,-0.2873265881396568,-0.04687250231793151,-0.1486856038310674,-0.23818844482467075,-0.18261554085175202,-0.1712691603702777,-0.16066017769279733,-0.24547983703137138,-0.17433931902975072,-0.1475700546295571,-0.24282206281908092,-0.17258733251847114,-0.20860483994139073,-0.2856003071660455,-0.24669107399883813,-0.19267565352579993,-0.23548467244145874,-0.28313204915917967,-0.30901747682969133,-0.3271171571343226,-0.37433808105229743,-0.31966228260249807,-0.403496487858844,-0.5589710348210146,-0.6038997192657387,-0.7003319954341806,-0.837923097009028,-0.7314617548376581,-0.6204005864775302,-0.5113223104766456,-0.5934726759293577,-0.5792066836556866,-0.6537543284126112,-0.521639696088781,-0.4217241593273932,-0.3979041062368499,-0.2910277507741779,-0.31653094056435865,-0.31271284702420393,-0.17190558801564598,-0.08957899475552873,-0.11081953284203906,-0.03474752513145364,-0.04273803850366873,-0.0945893286133175,-0.1092083831681716,-0.12328859725937506,-0.06884955708624804,-0.13204973448411933,-0.11285135007265497,-0.07342209572558546,-0.049653764389375445,-0.0060185542033113016,-0.012304230728715539,-0.03033137379449479,-0.0870801042477328,-0.051060493077810236,-0.09716156027806523,-0.09360847231298426,-0.021409110896949923,-0.019288562768526724,0.007279694880883179,0.03109176933235884,-0.003474912649559697,-0.04721441894953761,-0.10420688295944645,-0.125111532527651,-0.09964011781834058,-0.04368686304991073,0.03676178269467669,0.05320726680489388,-0.008322872148846333,0.0004626952428059367,-0.05966731863952908,-0.02806186225476696,0.0006996696958790644,0.0499386946349669,0.010813660107911276,-0.10732982981505824,-0.07139805397409209,-0.03921355006813629,0.009009420496005904,0.04099931120320911,0.0533418627986692,0.04253126040500153,0.06528986415156832,0.12228021356375059,0.15960670518015266,0.08372372982390724,0.006459690350206255,-0.01248786932106023,-0.03834896874779947,-0.05952857894653085,0.053137210651987336,0.19349836556919017,0.22723686795141687,0.24964362126010337,0.2992181947533352,0.2381853832373865,0.19781302925700686,0.28022986254133786,0.3639440049886739,0.49112866767507135,0.5565064194523442,0.4971357212400718,0.5076265238262139,0.39526762926183945,0.4302844776044973,0.420920889576361,0.24457877925128055,0.3038077879672332,0.32222883179434625,0.25225644609122055,0.2568586127366754,0.20205183857510467,0.09981149233248898,0.0417714935272594,0.0062164296319931565,-0.0026660564471514417,-0.11045485805656419,-0.09330518453446485,-0.1587807412988783,-0.12993409946838858,-0.2026265286213503,-0.2063194394120462,-0.31707495356896587,-0.3403044524109259,-0.265826726532376,-0.19467392586686375,-0.16763208606312263,-0.14764206352537496,-0.1261193812417531,-0.18396465488624775,-0.22590975411768877,-0.21706701333731385,-0.21283676080538105,-0.21546658386430362,-0.08517473296156217,-0.08397857457485829,-0.13418613751963562,-0.05665777656248427,-0.07465324024351906,-0.13348389243725253,-0.09047546847970046,-0.13089797866537128,-0.09940726543960572,-0.06422951178995304,-0.0183392833127454,-0.034132980967429546,-0.04899011572392867,-0.07714720444608565,-0.034806863601023204,-0.0036635745412776674,0.04104779263684857,0.17397405437771551,0.13982274977378295,0.12929718007609436,0.09876263748471063,0.15922211203792347,0.13467759741167148,0.22383216375027315,0.1979909654357368,0.11584274350362314,0.07906774074656625,0.04910099981924296,0.07243153259127154,0.15866013056379824,0.21790653548589348,0.1885144732949406,0.14468973309333996,0.14342334418882 [... cut ...]
      tf.predict(x_test, y_pred);
  

      Serial.print("Predicted proba are: ");
  
      for (int i = 0; i < 2; i++) {
          Serial.print(y_pred[i]);
          Serial.print(i == 1 ? '\n' : ',');
      }
 
      delay(1000);
    }
}
```

Actually, I don't change the `.ion` code, and only change the model code in keras python environment. But in Arduino serial monitor **Conv2D and Dense** doesn't work, **fully Dense** works. 

I guess it's the problem of input data stucture? or the TENSOR_ARENA_SIZE?

closed 2022-11-06 by whubaichuan

whubaichuan, 2022-11-07:
@eloquentarduino solved the problem. We should be careful about the SRAM limitation and TENSOR_ARENA_SIZE.