eloquentarduino/EloquentTinyML #51: Error while begining model

theloukou, 2022-11-10:
Hi. I'm a next to noob regarding ML stuff, but I am needed to deploy a model on ESP32.
The model I am given is a tflite file, converted to C array. Arena size is set according to arena estimator sizes. During the tf.begin() call, the serial prints are as such:

_Didn't find op for builtin opcode 'MUL' version '1'
Failed to get registration from op code  d
--->TF ERROR: Cannot allocate tensors_ 

From what i understand, the model has a mul (multiply) layer (it is the 1st one) that the esp32 code cannot find the equivalent op code. Thing is, the mul operation is supported by TFlite (the same is for all the rest of the layers of the model), so why is it that it cannot be found?

I can provide said model and device code if needed. Also, if this is more of a TFlite problem than an EloquentTinyML one, let me know to go ask elsewhere! :)

theloukou, 2022-11-11:
Following up my own issue, [TensorFlowLite_ESP32 version 0.9.0](https://github.com/tanakamasayuki/Arduino_TensorFlowLite_ESP32) indeed does not have MUL operation in it.
[TFLite_ESP32 has been updated to V1.0.0](https://github.com/tanakamasayuki/Arduino_TensorFlowLite_ESP32/releases/tag/1.0.0), which indeed does add the MUL operation, along with a lot others.
Are there any plans in integrating this newer version in EloquentTinyML?

eloquentarduino, 2022-11-26:
Hi @theloukou , thanks for pointing out that V1.0.0 is available, I'm very busy right now and I cannot track the changes. I'll take a look into integrating this version next week.

closed 2024-01-12 by eloquentarduino