eloquentarduino/EloquentTinyML #52: Error in output prediction of LSTM model with EloquentTinyML version 2.4.3

koulnodji6056, 2022-11-10:
Hi all,
I am trying to solve the issue but was not able since a month .
I do need your help to find out what is not working in order to move forward with my project.
I am trying to implement an LSTM model to predict temperature and humidity but the output is giving me **0** after all efforts
I have used xxd  to convert my tflite model to en h file since the tinymlgen kept giving me error.
libraries for model in colab: 
Python 3.7.15
TF version: 2.9.2
Keras: 2.9.0
Numpy: 1.21.6
Pandas: 1.3.5
implementation librairy:
Arduino IDE 2.0.0
 EloquentTinyML version 2.4.3
ESP 32 (dev module board)

**please help me find a solution**

below the link of the project :

https://github.com/koulnodji6056/Prediction_LSTM_model_EloquentTinyML_library

eloquentarduino, 2022-11-26:
I'm sorry, I'm having troubles replicating your project. If I use your sensor_model.h, I get "Only 1 subgraph is currently supported". Please add your complete Arduino sketch to your repository.

koulnodji6056, 2022-11-26:
@eloquentarduino Thanks for the reply.
 I have added the Arduino code.
Looking forward to hearing from you.

eloquentarduino, 2022-12-01:
If you replace `tf.begin(model)` with `while (!tf.begin(model)) Serial.print(tf.getErrorMessage());`, you will see that the error is `Only 1 subgraph supported`.

I don't know how to solve this error (even Googling didn't helped me). I think the best move is to upgrade to a later version of TensorFlow for Micro and see if they fixed this. Since I need time to do this, you can try using the latest version of this library (which mine internally uses) and report if it works for you.

https://github.com/tanakamasayuki/Arduino_TensorFlowLite_ESP32

koulnodji6056, 2022-12-01:
Thanks for the response. I will do that and let you know in case of any news.

eloquentarduino, 2022-12-06:
Spent many hours trying to fix this. Upgraded Tf in Python to 2.11, upgraded to latest Arduino_TensorFlowLite library, but cannot solve this problem. Did you find anything related to a solution?

koulnodji6056, 2022-12-06:
Unfortunately no. 
Still trying but unsuccessful