eloquentarduino/EloquentTinyML #53: Didn't find op for builtin opcode 'CONV_2D' version '5'   Failed to get registration from op code  d

Husnain-Bhatti, 2022-11-16:
I am trying to take an inference for a neural network on ESP 32 DEV KIT but it's throwing an error that is not understandable.

Please find my code below:

#include <EloquentTinyML.h>
#include <eloquent_tinyml/tensorflow.h>
#include "test_model.h"

 
#define NUMBER_OF_INPUTS 6
#define NUMBER_OF_OUTPUTS 1
#define TENSOR_ARENA_SIZE 2*1024

Eloquent::TinyML::TensorFlow::TensorFlow<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml;

void setup() {
    Serial.begin(115200);
    ml.begin(waterx_model);
}

void loop() {
    float x_test[7200] = {0.000000, 0.000139, ........ , 0.999722, 0.999861};
    float y_pred[4] = {0, 0, 0, 0};

    float predicted = ml.predict(x_test, y_pred);

    Serial.print("\t predicted: ");
    Serial.println(predicted);
    delay(10000);
}

below is my model summary:

[image]

eloquentarduino, 2022-11-17:
This kind of error is most times due to Tf version mismatch. Use Tf==2.2.4

Husnain-Bhatti, 2022-11-17:
Thanks, @eloquentarduino for your response. This model is working on TensorFlow 2.1.1 or below versions. But softmax is not working on these versions. If I don't apply softmax on the last dense layer it works fine but when I place softmax on the dense layer it throws error

abort() was called at PC 0x400de461 on core 1

Husnain-Bhatti, 2022-11-17:
@koulnodji6056 yes I have been able to do so. The issue occurs when I run it on ESP32 DEV KIT.

eloquentarduino, 2022-11-21:
Compatibility issues are pretty common with Tf, as you can see from other issues. I got replies from people who were able to deploy LSTM, so that's definetely possible. Please post the Python code so I can try to replicate.

koulnodji6056, 2022-11-21:
Thank you for the quick replay.
Here below is the link for the python code  and the dataset.
https://github.com/koulnodji6056/Prediction_LSTM_model_EloquentTinyML_library
kindly let me know if you need any additional information

koulnodji6056, 2022-11-21:
@eloquentarduino kindly share the link  from the people who were able to deploy LSTM, so that we can follow as well. thank you

eloquentarduino, 2022-11-26:
@koulnodji6056 I cannot find the links (if I remember correctly, nobody shared their code, they just said they got it working).

closed 2022-11-26 by eloquentarduino