eloquentarduino/EloquentTinyML #54: predict function predict(input, output) supports either int/int or float/float.?

Migue137, 2022-11-22:
WARNING: library Arduino_OV767X claims to run on mbed architecture(s) and may be incompatible with your current board which runs on mbed_nano architecture(s).
C:\Users\User\Documents\Cam Cris-copia\converted_tinyml\tm_template_script\cri_copi\cri_copi.ino: In function 'void loop()':
C:\Users\User\Documents\Cam Cris-copia\converted_tinyml\tm_template_script\cri_copi\cri_copi.ino:89:45: error: invalid conversion from 'tflite::ErrorReporter*' to 'int' [-fpermissive]
                             input->data.int8)) {
                                             ^
C:\Users\User\Documents\Cam Cris-copia\converted_tinyml\tm_template_script\cri_copi\cri_copi.ino:89:45: error: invalid conversion from 'int' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
C:\Users\User\Documents\Cam Cris-copia\converted_tinyml\tm_template_script\cri_copi\cri_copi.ino:89:45: error: too many arguments to function 'bool GetImage(int, int, int, uint8_t*)'
In file included from C:\Users\User\Documents\Cam Cris-copia\converted_tinyml\tm_template_script\cri_copi\cri_copi.ino:6:0:
C:\Users\User\Documents\Cam Cris-copia\converted_tinyml\tm_template_script\cri_copi\image_provider.h:21:6: note: declared here
 bool GetImage(int image_width, int image_height, int channels, uint8_t* image_data);
      ^~~~~~~~

exit status 1

Compilation error: invalid conversion from 'tflite::ErrorReporter*' to 'int' [-fpermissive]

When I try to compile the program i get this error, but in the issues section i already see that a user face up something similar, especifically the issue number 34.
In this issue you mention "I will fix the library on this. For now, declare your input as float so you can use the float/float version of predict.", and I want to know if you already solved it or not.
Or maybe you can tell me how to proceed.

eloquentarduino, 2022-11-26:
No, I didn't solved. Tell me what input and output datatypes you need.

Migue137, 2022-11-29:
I am not really sure, I was looking the issues section and I think that with that part that you mantion I can solve my problem.
First I will share an image so you can understand it better.

[image]

I also see that there is a matrix where you put 4 values but when you call it you add an extra value named error-probider, I think that maybe this is causing the error.
But I do not understand it very well.

eloquentarduino, 2022-12-01:
Please send the whole sketch.

Migue137, 2022-12-05:
Here is the link for the arduino sketch, it was too big so I had to put it in google drive

https://drive.google.com/file/d/1nMQvWksDtZUeHM3pnOaj9i2EQ8XC2tW6/view?usp=share_link

eloquentarduino, 2022-12-06:
You are not using the EloquentTinyML library. Why did you opened an issue here? You will have better luck asking for help on the TensorFlow forum. From the screen you sent, it is a syntax error, but I see you commented out that line, right?