eloquentarduino/EloquentTinyML #58: TfLite' in namespace 'Eloquent::TinyML' does not name a template type

0oK3rn3lP4n1co0, 2023-03-03:
Hi I have a little problem compiling.
I have already tried the version 2.4.4 to 2.4.1 of the lib.
I tried to compile it in the Arduino Ide and in the arduino-cli.
I tried it with my own code and also with this example code.

```
#include <EloquentTinyML.h>
#include <eloquent_tinyml/tensorflow.h>
// sine_model.h contains the array you exported from the previous step with xxd or tinymlgen
#include "sine_model.h"

#define NUMBER_OF_INPUTS 1
#define NUMBER_OF_OUTPUTS 1
// in future projects you may need to tweak this value: it's a trial and error process
#define TENSOR_ARENA_SIZE 2*1024

Eloquent::TinyML::TfLite<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml;

void setup() {
    Serial.begin(115200);
    ml.begin(sine_model);
}

void loop() {
    for (float i = 0; i < 10; i++) {
        // pick x from 0 to PI
        float x = 3.14 * i / 10;
        float y = sin(x);
        float input[1] = { x };
        float predicted = ml.predict(input);

        Serial.print("sin(");
        Serial.print(x);
        Serial.print(") = ");
        Serial.print(y);
        Serial.print("\t predicted: ");
        Serial.println(predicted);
        delay(1000);
    }
}
```
I get this error output from the compiler.
```
tensor_flow_test:23:19: error: 'TfLite' in namespace 'Eloquent::TinyML' does not name a template type
 Eloquent::TinyML::TfLite<NUMBER_OF_INPUTS, NUMBER_OF_OUTPUTS, TENSOR_ARENA_SIZE> ml;
                   ^~~~~~
Bibliothek EloquentTinyML in Version 2.4.4 im Ordner: /home/k3rn3lp4n1c/Arduino/libraries/EloquentTinyML  wird verwendet
exit status 1
'TfLite' in namespace 'Eloquent::TinyML' does not name a template type
```
What is the best way to solve this problem?

Thanks a lot

maxall41, 2023-04-10:
Im having the same issue

maxall41, 2023-04-10:
Update: It seems to work if you use the examples under "2.4.0" instead of the normal examples

eloquentarduino, 2023-12-07:
If using ESP32, move to https://github.com/eloquentarduino/EloquentTensorFlow32

closed 2023-12-07 by eloquentarduino