eloquentarduino/EloquentTinyML #59: There is a problem with this program code, sir.     EloquentTinyML/examples/2.4/PersonDetection/PersonDetection.ino

331319341, 2023-04-24:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (11565) cam_hal: FB-SIZE: 0 != 76800
No person detected
	 > It took 5134ms to detect
	 > Person score: 155
	 > Not person score: 170
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d202b  PS      : 0x00060a30  A0      : 0x800d23aa  A1      : 0x3ffb27e0  
A2      : 0x00000000  A3      : 0x00000003  A4      : 0x0000000a  A5      : 0x3ffd8230  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x3ffd92f8  A9      : 0x3ffb27c0  
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x0000000a  A13     : 0x0000001d  
A14     : 0x3ffb30d8  A15     : 0x25211f1d  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40087fc5  LEND    : 0x40087fd5  LCOUNT  : 0xfffffffe      

The program can only run once and then restarts.

331319341, 2023-04-24:
EloquentTinyML/examples/2.4/Esp32PersonDetection/      https://github.com/eloquentarduino/EloquentTinyML/tree/master/examples/2.4/Esp32PersonDetection   
[image]
This project includes a missing package, so my code can't be executed no matter what I do.

eloquentarduino, 2024-01-12:
Please refer to the updated tutorial: https://eloquentarduino.com/posts/esp32-cam-person-detection

closed 2024-01-12 by eloquentarduino