eloquentarduino/EloquentTinyML #68: Error compiling for Teensy 4.1

SebyR, 2023-10-31:
Hello, when I verify the code with the Teensy 4.1 board selected I get the following error:
In file included from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h: In function 'T tflite::FloatToQuantizedType(float, float, int)':
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:39:42: error: 'numeric_limits' is not a member of 'std'
   39 |       std::max(static_cast<int32_t>(std::numeric_limits<T>::min()), result);
      |                                          ^~~~~~~~~~~~~~
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:39:58: error: expected primary-expression before '>' token
   39 |       std::max(static_cast<int32_t>(std::numeric_limits<T>::min()), result);
      |                                                          ^
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:39:61: error: '::min' has not been declared; did you mean 'std::min'?
   39 |       std::max(static_cast<int32_t>(std::numeric_limits<T>::min()), result);
      |                                                             ^~~
      |                                                             std::min
In file included from c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\algorithm:62,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:20,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\bits\stl_algo.h:3455:5: note: 'std::min' declared here
 3455 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
In file included from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:41:42: error: 'numeric_limits' is not a member of 'std'
   41 |       std::min(static_cast<int32_t>(std::numeric_limits<T>::max()), result);
      |                                          ^~~~~~~~~~~~~~
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:41:58: error: expected primary-expression before '>' token
   41 |       std::min(static_cast<int32_t>(std::numeric_limits<T>::max()), result);
      |                                                          ^
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:41:61: error: '::max' has not been declared; did you mean 'std::max'?
   41 |       std::min(static_cast<int32_t>(std::numeric_limits<T>::max()), result);
      |                                                             ^~~
      |                                                             std::max
In file included from c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\algorithm:62,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:20,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\bits\stl_algo.h:3467:5: note: 'std::max' declared here
 3467 |     max(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
In file included from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h: In function 'T tflite::FloatToSymmetricQuantizedType(float, float)':
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:49:42: error: 'numeric_limits' is not a member of 'std'
   49 |       std::max(static_cast<int32_t>(std::numeric_limits<T>::min() + 1), result);
      |                                          ^~~~~~~~~~~~~~
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:49:58: error: expected primary-expression before '>' token
   49 |       std::max(static_cast<int32_t>(std::numeric_limits<T>::min() + 1), result);
      |                                                          ^
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:49:61: error: '::min' has not been declared; did you mean 'std::min'?
   49 |       std::max(static_cast<int32_t>(std::numeric_limits<T>::min() + 1), result);
      |                                                             ^~~
      |                                                             std::min
In file included from c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\algorithm:62,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:20,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\bits\stl_algo.h:3455:5: note: 'std::min' declared here
 3455 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~
In file included from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:51:42: error: 'numeric_limits' is not a member of 'std'
   51 |       std::min(static_cast<int32_t>(std::numeric_limits<T>::max()), result);
      |                                          ^~~~~~~~~~~~~~
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:51:58: error: expected primary-expression before '>' token
   51 |       std::min(static_cast<int32_t>(std::numeric_limits<T>::max()), result);
      |                                                          ^
c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:51:61: error: '::max' has not been declared; did you mean 'std::max'?
   51 |       std::min(static_cast<int32_t>(std::numeric_limits<T>::max()), result);
      |                                                             ^~~
      |                                                             std::max
In file included from c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\algorithm:62,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:20,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\bits\stl_algo.h:3467:5: note: 'std::max' declared here
 3467 |     max(initializer_list<_Tp> __l, _Compare __comp)
[... 28 lines cut ...]
      |                                                     ^~~
[... cut ...]
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src/eloquent_tinyml/tensorflow/arm/tensorflow/lite/micro/micro_utils.h:20,
                 from c:\Users\Sebi\Documents\Arduino\libraries\EloquentTinyML\src\eloquent_tinyml\tensorflow\arm\tensorflow\lite\micro\micro_utils.cpp:17:
c:\users\sebi\appdata\local\arduino15\packages\teensy\tools\teensy-compile\11.3.1\arm\arm-none-eabi\include\c++\11.3.1\bits\stl_algo.h:3455:5: note: 'std::min' declared here
 3455 |     min(initializer_list<_Tp> __l, _Compare __comp)
      |     ^~~

vladBaciu, 2023-11-04:
Just compile with the flag -D TF_LITE_USE_GLOBAL_MIN

SebyR, 2023-11-04:
> Just compile with the flag -D TF_LITE_USE_GLOBAL_MIN set

Where should I insert the flag? P.S I am also Romanian but i will write in English

vladBaciu, 2023-11-04:
How do you compile the files more exactly? If you just do it from Arduino IDE, better to this inside min.h and max.h files:

#if defined(TF_LITE_USE_GLOBAL_MIN) || defined(__ZEPHYR__) **|| defined(ARDUINO)**
#if defined(TF_LITE_USE_GLOBAL_MAX) || defined(__ZEPHYR__) **|| defined(ARDUINO)**

 I think the latest TF package has updated the tflite files as above.

SebyR, 2023-11-05:
> How do you compile the files more exactly? If you just do it from Arduino IDE, better do this inside min.h and max.h files:
> 
> #if defined(TF_LITE_USE_GLOBAL_MIN) || defined(**ZEPHYR**) _**|| defined(ARDUINO)**_ #if defined(TF_LITE_USE_GLOBAL_MAX) || defined(**ZEPHYR**) _**|| defined(ARDUINO)**_
> 
> I think the latest TF package has updated the code as above.

I am using Arduino IDE. I modified the eloquent library where you sugested but no succes, it still shows the same error.

vladBaciu, 2023-11-28:
Try to add in micro_utils.h `#include <limits>`

eloquentarduino, 2024-01-12:
I don't use the Teensy board (although I have one). Do you use Arduino IDE or Teensyduino IDE?

SebyR, 2024-02-04:
> I don't use the Teensy board (although I have one). Do you use Arduino IDE or Teensyduino IDE?

Sorry for the late answer, I had a lot of schoolwork to do. I am using Arduino IDE with a Teensy 4.1 board. It was a problem with only the 2.4.0 library. When I used one of the older libraries it worked. Now I will try to use the 3.0.1 library, but I will remake the training because the model (the one that I used with the 2.4.0 library) isn't compatible anymore.

closed 2024-02-04 by SebyR