eloquentarduino/EloquentTinyML #71: What the meaning of TF_NUM_OPS ？

YuriSizuku, 2024-01-22:
I try to read the source code but don't understand this parameter. 
Could you please explain or add some comments about this parameter ?

jakobkilian, 2024-01-24:
I would also be very interested in this. Unfortunately, I can't get my models to run with the update, but I don't know what should be entered there...
Thanks for the great work, Simone!

jakobkilian, 2024-01-24:
Didn't really solve this but here is my research:
- TF_NUM_OPS eventually will go [here](https://github.com/tensorflow/tflite-micro/blob/324ae1eaba98c8aeb440072371efea481c42ace3/tensorflow/lite/micro/micro_mutable_op_resolver.h#L45) as `tOpCount`
- I am not experienced enough to explain it fully but by reading [this](https://github.com/tensorflow/tflite-micro/blob/324ae1eaba98c8aeb440072371efea481c42ace3/tensorflow/lite/micro/micro_mutable_op_resolver.h#L45) it seems to be the number of "operations" that you add later by e.g. doing: `tf.resolver.AddFullyConnected();` or `tf.resolver.AddSoftmax();`

So, if you only add FullyConnected TF_NUM_OPS should be 1? I have the feeling this library here will be dropped anyways in favour of [this](https://github.com/eloquentarduino/EloquentTensorFlow32) one, which also uses the number of operations as argument....

YuriSizuku, 2024-01-24:
> Didn't really solve this but here is my research:
>
> * TF_NUM_OPS eventually will go [...]

Thank you for the information. 
Is this the number of operations ?  Is that means we needs to pre allocate memory before using each operation ?

jakobkilian, 2024-01-26:
Just a short note as I understand some more things now and may be able to explain this partly:

There is a [couple of operations in tflite ](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_mutable_op_resolver.h). With `TF_NUM_OPS` in this library (and also in others) you just reserve memory to do these operations. 

Example: 
When using a quantized model (taking in float, quantizing it to an int, running the inference and dequantize it back to a float) you would need `TF_NUM_OPS` to be at least 3 because:

```cpp
#define TF_NUM_OPS 3
#define ARENA_SIZE 2000
Eloquent::TF::Sequential<TF_NUM_OPS, ARENA_SIZE> tf;

void setup() {
    //stuff
    tf.resolver.AddFullyConnected();
    tf.resolver.AddQuantize();
    tf.resolver.AddDequantize();
}
```

with this you could load a quantized model like described [here in the end](https://www.oreilly.com/library/view/tinyml/9781492052036/ch04.html)

YuriSizuku, 2024-01-28:
> Just a short note as I understand some more things now and may be able to explain this partly:
> 
> There is a [couple of possible operations in tflite](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_mutable_op_resolver.h). With `TF_NUM_OPS` in this library (and also in others) you just reserve memory to do these operations.
> 
> Example: When using a quantized model (taking in float, quantizing it to an int, running the inference and dequantize it back to a float) you would need `TF_NUM_OPS` to be at least 3 because:
> 
> ```c++
> #define TF_NUM_OPS 3
> #define ARENA_SIZE 2000
> Eloquent::TF::Sequential<TF_NUM_OPS, ARENA_SIZE> tf;
> 
> void setup() {
>     //stuff
>     tf.resolver.AddFullyConnected();
>     tf.resolver.AddQuantize();
>     tf.resolver.AddDequantize();
> }
> ```
> 
> with this you could load a quantized model like described [here in the end](https://www.oreilly.com/library/view/tinyml/9781492052036/ch04.html)

Thank you for your explanation. I wonder how do we know what operations are necessory for the model ?

jakobkilian, 2024-01-28:
I also wonder why there seems to be no good docu about it.
If you train your model yourself, you can decide on its architecture and the operations you build in. When you have a tflite file and no knowledge you can use [Netron](https://netron.app/) to analyse the model. This is e.g. a tiny [quantized tflite model] (https://github.com/eloquentarduino/EloquentTinyML/files/14076051/sine_model_quantized.tflite.zip) I trained as a test. It has quantization built in as you can see in the screenshot and there is a correspinding `AddFullyConnected` in the mentioned [list](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_mutable_op_resolver.h).
Still I don't fully understand this: this model runs for example even if you just add a single `AddFullyConnected()` and `AddQuantize()` only (not multiple fully connected and dequantize).

YuriSizuku, 2024-01-30:
> > I also wonder why there seems to be no good docu about it. If you train your model yourself, you can decide on its architecture and the operations you build in. When you have a tflite file and no knowledge you can use [Netron](https://netron.app/) to analyse the model. This is e.g. a tiny [quantized tflite model](https://github.com/eloquentarduino/EloquentTinyML/files/14076051/sine_model_quantized.tflite.zip) I trained as a test. It has quantization built in as you can see in the screenshot and there is a correspinding `AddFullyConnected` in the mentioned [list](https://github.com/tensorflow/tflite-micro/blob/main/tensorflow/lite/micro/micro_mutable_op_resolver.h). Still I don't fully understand this: this model runs for example even if you just add a single `AddFullyConnected()` and `AddQuantize()` only (not multiple fully connected and dequantize).
> > ![Netron App Screenshot](https://private-user-images.githubusercontent.com/46564824/300263603-8fc3a083-f5c8-4a4b-a5dd-1ea496213687.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY2MTc0NzgsIm5iZiI6MTcwNjYxNzE3OCwicGF0aCI6Ii80NjU2NDgyNC8zMDAyNjM2MDMtOGZjM2EwODMtZjVjOC00YTRiLWE1ZGQtMWVhNDk2MjEzNjg3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTMwVDEyMTkzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU3NTY1MGU3OGEyNzBhNjZiMGJmNzZkODc4MzZkZjI3ZjVhOTUxNGE0ODIzYTgwOWNlYjJkODZiZjE0NjBlNGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7haZF4yPhUP1Y1-_G5rhMvmOn3bF9Sk3z_49KugEAvo)
> 
> I found a easy way to summary the ops in a tflite model. `python -m tensorflow.lite.tools.visualize xxx.tflite `