eloquentarduino/EloquentTinyML #73: issue with deploying model on Sony Spresense using the EloquentTinyML

Abdullah-Shahid01, 2024-08-01:
Code:
#include "irisModel.h"
#include <tflm_cortexm.h>
#include <eloquent_tinyml.h>
#define ARENA_SIZE 10000

Eloquent::TF::Sequential<TF_NUM_OPS, ARENA_SIZE> tf;

void setup() {
  Serial.begin(115200);
  delay(3000);
  Serial.println("__TENSORFLOW IRIS__");

  // configure input/output
  // (not mandatory if you generated the .h model
  // using the everywhereml Python package)
  tf.setNumInputs(4);
  tf.setNumOutputs(3);
  // add required ops
  // (not mandatory if you generated the .h model
  // using the everywhereml Python package)
  tf.resolver.AddFullyConnected();
  tf.resolver.AddSoftmax();

  while (!tf.begin(irisModel).isOk()){
    Serial.println(tf.exception.toString());
  }

}

void loop() {
  // classify class 0
    if (!tf.predict(x0).isOk()) {
        Serial.println(tf.exception.toString());
        return;
    }
    
    Serial.print("expcted class 0, predicted class ");
    Serial.println(tf.classification);
    
    // classify class 1
    if (!tf.predict(x1).isOk()) {
        Serial.println(tf.exception.toString());
        return;
    }
    
    Serial.print("expcted class 1, predicted class ");
    Serial.println(tf.classification);
    
    // classify class 2
    if (!tf.predict(x2).isOk()) {
        Serial.println(tf.exception.toString());
        return;
    }
    
    Serial.print("expcted class 2, predicted class ");
    Serial.println(tf.classification);

    // how long does it take to run a single prediction?
    Serial.print("It takes ");
    Serial.print(tf.benchmark.microseconds());
    Serial.println("us for a single prediction");
    
    delay(1000);

}

Error:
/Arduino/libraries/tflm_cortexm/src/tensorflow/lite/micro/system_setup.cpp:68:39: error: expected template-name before '<' token
   68 | class _RingBuffer : public RingBufferN<kSerialMaxInputLength + 1> {
      |                                       ^

/Arduino/libraries/tflm_cortexm/src/tensorflow/lite/micro/system_setup.cpp:68:39: error: expected '{' before '<' token
/Arduino/libraries/tflm_cortexm/src/tensorflow/lite/micro/system_setup.cpp:68:39: error: expected unqualified-id before '<' token
/Arduino/libraries/tflm_cortexm/src/tensorflow/lite/micro/system_setup.cpp:73:20: error: aggregate 'test_over_serial::_RingBuffer test_over_serial::_ring_buffer' has incomplete type and cannot be defined
   73 | static _RingBuffer _ring_buffer;
      |                    ^~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

eloquentarduino, 2024-08-13:
I don't have this board and cannot test. I see it has a Cortex M4F CPU, which should be similar to Arduino Nano 33 BLE. I will try to investigate on this board instead.

Inmoresentum, 2024-08-13:
:)

closed 2024-08-26 by eloquentarduino