eloquentarduino/EloquentTinyML #76: how to modify model parameters

arpit-softcircuits, 2024-12-27:
i am trying to run some ML model on ESP32. i have generated Model.h file using google colab.
this is my model 

img_height=64
img_width=64

model = Sequential([
  layers.Rescaling(1./255, input_shape=(img_height, img_width, 1)),
  layers.Conv2D(16, 3, padding='same', activation='relu'),
  layers.MaxPooling2D(),
  layers.Conv2D(32, 3, padding='same', activation='relu'),
  layers.MaxPooling2D(),
  layers.Conv2D(64, 3, padding='same', activation='relu'),
  layers.MaxPooling2D(),
  layers.Flatten(),
  layers.Dense(128, activation='relu'),
  layers.Dense(num_classes)
])

 how to set values of following
    tf.setNumInputs(?);
    tf.setNumOutputs(?);
    tf.resolver.Add<?>;
    tf.resolver.Add<?>;

arpit-softcircuits, 2024-12-27:
i guss 
  NumInputs=img_height * img_width
  NumOutputs=num_classes

let me know if i am correct?
but i am confused about 
       TF_NUM_OPS = ?
and what resolvers need to be add?

eloquentarduino, 2024-12-28:
TF_NUM_OPS is the number of distinct operators (dense, flatten, Conv2d...). If not sure, use 10 for your network.

You have to add to the resolver all the layers you are using. This is an example for a LSTM model:

`nn.resolver.AddPack();
    nn.resolver.AddSplit();
    nn.resolver.AddTranspose();
    nn.resolver.AddMul();
    nn.resolver.AddGather();
    nn.resolver.AddMinimum();
    nn.resolver.AddSlice();
    nn.resolver.AddWhile();
    nn.resolver.AddReshape();
    nn.resolver.AddShape();
    nn.resolver.AddFill();
    nn.resolver.AddSoftmax();
    nn.resolver.AddStridedSlice();
    nn.resolver.AddTanh();
    nn.resolver.AddLess();
    nn.resolver.AddMaximum();
    nn.resolver.AddAdd();
    nn.resolver.AddFullyConnected();
    nn.resolver.AddUnidirectionalSequenceLSTM();
    nn.resolver.AddRelu();
    nn.resolver.AddConcatenation();
    `

In your case you will have to add Conv2D, MaxPool2D, Flatten, Dense. I don't know if rescaling is supported.

Here's the full list of layers available: https://github.com/tanakamasayuki/Arduino_TensorFlowLite_ESP32/blob/e88e0ebee0430ed716ff5b49854795db90066e59/src/tensorflow/lite/micro/micro_mutable_op_resolver.h#L40